ECSHackWeek/impedance.py #13: generate quality circuit diagrams

mdmurbach, 2018-10-06:
Have had a few requests for the ability to export a circuit to a pdf/svg/png 

After some poking around I think SchemDraw (https://cdelker.bitbucket.io/SchemDraw/SchemDraw.html) seems like a good way to do it

mdmurbach, 2019-05-20:
scbarton got a pretty sweet draw_circuit function started in #25. Testing more complex functions and documentation could be a great first contributor/Hack Week project.

dacb, 2020-02-26:
I have code from our last hack week that mostly implements this.  One of the problems is that not all elements are represented in the circuit library in SchemDraw, IIRC.  I'll try to make time to review this over the next week.  It is also against a nearly one year old fork so that will need to reconciled.

mdmurbach, 2020-02-27:
Great! Let me know if/how I can help

beckdac, 2020-02-27:
Writing good tests is going to be a challenge.

mdmurbach, 2020-02-27:
Yeah... I could see that.

I think it'd be possible to come up with some edge cases (nested parallel/series elements, etc.) but testing/ensuring that it works globally seems challenging. Maybe we just check the circuit complexity and warn the user if it's beyond a typical (i.e. tested) use case. Realistically, there are physical constraints on what complexity of circuit makes sense if that's helpful.

Yugal0, 2023-08-19:
The circuit drawing feature is not yet there in the library, right?
or it is that I have an older distribution,
Also, can someone tell me what the schedule is for releasing a new distribution of the library on pip (the python package installer).