ECSHackWeek/impedance.py #19: example notebook fitting_example.ipynb calls outdated version of impedance.plotting.plot_nyquist func

hkennyv, 2019-01-16:
Hello,

Came across this minor documentation bug when running the `fitting_example.ipynb` notebook. 

this block of code passes an outdated argument to the `plot_nyquist` function. 
```
plot_nyquist(ax, f_pred, randles_fit, fmt='-')
plot_nyquist(ax, f_pred, randlesCPE_fit, fmt='-')
plot_nyquist(ax, f_pred, customCircuit_fit, fmt='-')
```

It should be corrected to this to achieve the same behavior:
```
plot_nyquist(ax, f_pred, randles_fit, fit=True)
plot_nyquist(ax, f_pred, randlesCPE_fit, fit=True)
plot_nyquist(ax, f_pred, customCircuit_fit, fit=True)
```

I'm a relatively new contributor, but I'd gladly submit a PR if you could tell me which branch to submit it to. I noticed the master branch's source code still uses the old `fmt` parameter, but the package on PyPi uses the new parameter.

mdmurbach, 2019-01-16:
Hi Kenny,

Thanks for pointing this out. The master branch is actually the more up to date version of the code, so the `plot_nyquist(ax, f_pred, randles_fit, fmt='-')` behavior is actually the newer version relative to the v0.1.1 release on PyPI.

To get around this you can clone and use the development version by cloning the git repository and installing directly:

```
git clone https://github.com/ECSHackWeek/impedance.py

cd impedance.py
python setup.py install --user
```

There is a [PR under review currently](https://github.com/ECSHackWeek/impedance.py/pull/17) and we will be releasing another version after that get merged which will allow the updated version to be installed via pip.

Thanks and have a good afternoon,
  -- Matt

hkennyv, 2019-01-16:
Great! Will do, thank you Matt.

closed 2019-01-16 by hkennyv