ECSHackWeek/impedance.py #27: Better checking/handling of custom circuit string definitions

mdmurbach, 2019-05-30:
The error that gets thrown if an incorrect circuit is specified is not intuitive (the error is raised when the number of initial guesses doesn't match the circuit). For example,

```
custom = CustomCircuit('F0-T0', initial_guess=[1, 2, 3, 4, 5])
```

raises `AssertionError: Initial guess length needs to be equal to {circuit_length}`.

We could add a check during initialization that all the elements specified are allowed and raise a more useful error at that time.

nealde, 2020-02-21:
While we're in there validating the circuit, we should decide if we want to enforce strict numbering - i.e. does R2 have to follow R1? is R1-R3-R2 acceptable, or even R1-R3?

lktsui, 2020-02-29:
Regarding @nealde's comment, I think different groups have different standards of naming their elements. I tend to name mine elements grouped in parallel units since R1 and C1 correspond to one electrochemical process and R2/L2/C2 correspond to some other process, so all the 1s go together and all the 2s go together.

`R0-(R1/C1)-(R2/L2/C2)`

but you could certainly name it 

`R0-(R1/C0)-(R2/L0/C1)`

I think as long as each element is uniquely named, I would like to err on the side of giving the user flexibility in defining how they want to name their circuit elements.

closed 2020-06-06 by mdmurbach