ECSHackWeek/impedance.py #29: Is it possible to fix a value in the fit?

michaelplews, 2019-06-18:
**Is your feature request related to a problem? Please describe.**
When I guess a value, I need to give a range. If the range is too small, the fit takes a long time. There are certain values I want to fix in the fit, but can't see an option to do so.

**Describe the solution you'd like**
By providing a fixed value, the fit should take less time (I think??)

**Describe alternatives you've considered**
A small boundary works, but is still computationally intensive.

Is there currently a way to do this that I can't find? Thanks!

mdmurbach, 2019-06-18:
Unfortunately, there is currently not a way to easily do this, although it's definitely a feature I think would be useful.

The fitting in impedance.py is currently done using [scipy.optimize.curve_fit](https://docs.scipy.org/doc/scipy/reference/generated/scipy.optimize.curve_fit.html) which doesn't offer an option for setting a parameter. Off the top of my head, I can think of a few options for adding this kind of functionality. 

1. Switching to a different fitting package ([lmfit](https://lmfit.github.io/lmfit-py/) seems like an interesting option) that allows parameter values to be fixed. We then would have to figure out how to write the fitting function using the Parameter objects in lmfit and allow these parameters to be set in the circuit object.

2. Build a wrapper function around the circuit fitting function that would allow parameters to be fixed while still relying on curve_fit for the actual fitting. 

Happy to hear your/others thoughts on the value of this feature and/or suggestions for a pathway towards implementing it. Thanks!

mdmurbach, 2019-09-05:
Implemented the functionality for 2. above in PR #36.

You can now define a circuit as

```python
randles = Randles(initial_guess=[None, .005, .1, .001, 200], constants={'R0': 0.01))
``` 

or
```python
customConstantCircuit = CustomCircuit(initial_guess=[None, .005, .1, .005, .1, .001, None], constants={'R0': 0.02, 'W1_1': 200}, circuit='R0-p(R1,C1)-p(R2,C2)-W1')
```

mdmurbach, 2019-09-09:
This was added in v0.4!

You can update with `pip install --upgrade impedance`. Please let me know if you run into any issues with the new feature :smile:

closed 2019-09-09 by mdmurbach

mdmurbach, 2019-10-21:
@all-contributors please add @michaelplews for ideas