ECSHackWeek/impedance.py #30: Importing impedance.circuits[BUG]

Chebuskin, 2019-07-03:
**Describe the bug**
When importing .circuits no display is found and it starts using the Agg backend. Thus, you can not show graphs with matplotlib

**To Reproduce**
Steps to reproduce the behavior:
1. Code which generates the error '...'
import impedance.circuits
2. Full error info
no display found. Using non-interactive Agg backend

I was just trying to try the fitting_example.ipynb but ran into this error and discovered it was occuring with the .circuits import

mdmurbach, 2019-07-10:
Hi @Chebuskin,

I'm not able to reproduce your issue so far, can you tell us a little bit more about your:

- operating system
- python version (`python -V`)
- matplotlib and jupyter lab versions (`pip freeze`)
- how you installed impedance.py

It seems like you might be missing a back end that works with the jupyter notebook/lab you're working with. Maybe try creating a new virtual environment with conda and trying in there ([see here for an example](https://impedancepy.readthedocs.io/en/latest/getting-started.html))?

Thanks!

mdmurbach, 2019-07-10:
If you are running the code over ssh, you may also want to try the `-X` option ([see here for example](https://stackoverflow.com/a/9303165)) to correctly set the DISPLAY variable and avoid setting the backend to Agg.

mdmurbach, 2019-08-14:
This looks like it could be the same issue as #32. Should be fixed in [v0.3.1](https://pypi.org/project/impedance/0.3.1/).

You can upgrade with `pip install --upgrade impedance`. Please let me know if you're still running into any issues!

closed 2019-08-14 by mdmurbach

mdmurbach, 2019-10-21:
@all-contributors please add @Chebuskin for bug