ECSHackWeek/impedance.py #31: Units of Capacitance?

environmat, 2019-07-19:
First, thanks for the great work and providing such a cool tool! 

Second, with the potential on looking stupid, I did not figure out the units of the capacitance yet. Is it F, µF or what is it? 

Thanks, 
Matthias

lucasCFO, 2019-08-19:
Yeah, it would be great to see the units when calling print(circuit) !

Thanks

mdmurbach, 2019-08-20:
Hi @environmat and @lucasCFO, my apologies for missing this earlier! Capacitance is just in F and I agree that it would be great to print the units alongside the initial guesses/parameters. It looks like we could add a decorator to each circuit element with a unit field similar to how we store the number of parameters. Happy to review a pull request or other ideas!

mdmurbach, 2019-09-09:
This was added in the v0.4 release!

You can update with `pip install --upgrade impedance` :smile:

closed 2019-09-09 by mdmurbach

mdmurbach, 2019-10-21:
@all-contributors please add @environmat for bug