ECSHackWeek/impedance.py #32: no display found. Using non-interactive Agg backend on windows

abdullahsumbal, 2019-08-14:
**Describe the bug**
First of all, great work with the impedance package. When I import `Randles` from `impedance.circuits`, it does not let me plot figures using matplotlib. The package is checking environment variable `DISPLAY` in [curcuits.py](https://github.com/ECSHackWeek/impedance.py/blob/7c81b149890f62493fddb928913583bdc9a3a6aa/impedance/circuits.py#L10). I think windows (I am a window user) does not have a `DISPLAY` environment variable, atleast mine does not. Thats why in [circuits.py](https://github.com/ECSHackWeek/impedance.py/blob/7c81b149890f62493fddb928913583bdc9a3a6aa/impedance/circuits.py#L12), line `mpl.use('Agg')` is executed. There are some potential solutions to check for display in windows in this [post](https://stackoverflow.com/a/45756291/11567320)  

**To Reproduce**
Steps to reproduce the behavior:
1. Code which generates the error 
```
import matplotlib.pyplot as plt
from impedance.circuits import Randles

plt.plot([1, 2, 3, 4])
plt.show()
....
```
2. Full error info
I am getting the following message when I am importing `Randles` from `impedance.circuits`.
```
no display found. Using non-interactive Agg backend
C:\Users\Sumbal\Documents\atom\data-analysis\antranik\controller\controller.py:193: UserWarning: Matplotlib is currently using agg, which is a non-GUI backend, so cannot show the figure.
  plt.show()
```

**Expected behavior**
This message above should not appear.

**Additional context**
* how I installed
`pip install impedance==0.3.0`
* operating system
`windows 10`

mdmurbach, 2019-08-14:
Hi Abdullah,

Thanks for taking the time to report this. I think the best way to address this is probably to remove the display detection lines https://github.com/ECSHackWeek/impedance.py/blob/7c81b149890f62493fddb928913583bdc9a3a6aa/impedance/circuits.py#L10 meant to detect when the package was being used over the network or headless (i.e. travis ci). It looks like setting the display variable in `.travis.yml` instead [should now work](https://docs.travis-ci.com/user/gui-and-headless-browsers/#using-services-xvfb). PR to check if this works incoming...

mdmurbach, 2019-08-14:
This should be fixed in [v0.3.1](https://pypi.org/project/impedance/0.3.1/)! 

You can update with `pip install --upgrade impedance`. Please let me know if you still run into any issues :smile:

closed 2019-08-14 by mdmurbach

abdullahsumbal, 2019-08-22:
Update fixed the issue. Thanks a lot. 👍

mdmurbach, 2019-10-21:
@all-contributors please add @abdullahsumbal for bug