ECSHackWeek/impedance.py #37: [Documentation] Gerischer element documentation unclear

BGerwe, 2019-10-08:
The documentation for the Gerischer element is:
[image]

But in circuit_elements.py the definition of this element is:
[image]
The documentation should be updated to reflect the code.

Alternatively, this element could be defined following Lu, Kreller, and Adler, 2009 (DOI: 10.1149/1.3079337) :
[image]
With equivalence between parameters as: 
[image]

The benefit of this definition is R_G is interpretable as the impedance width (rather than an admittance as is the case with Y), and t_G is a characteristic time constant where the system begins transitioning from co-limited kinetics and diffusion to only diffusionally limited.

mdmurbach, 2019-10-08:
Awesome, thanks for taking the time to write this up! :1st_place_medal: 

> The benefit of this definition is R_G is interpretable as the impedance width (rather than an admittance as is the case with Y), and t_G is a characteristic time constant where the system begins transitioning from co-limited kinetics and diffusion to only diffusionally limited.

I definitely would favor the more meaningful parameter interpretations, so I think it makes sense to redefine the element (isn't that the point of a version <1.0/beta anyways :wink:). Would you want to make these changes and submit a pull request? Happy to help answer any questions you might have along the way.

BGerwe, 2019-10-09:
I think I've made the appropriate changes, but I'm not sure how to check how they'll appear rendered on the docs page. Also, this is my first time pushing to a repo owned by someone else, so I'm a little unclear if I need permissions and how exactly I should branch it.

mdmurbach, 2019-10-09:
Awesome! :rocket: 

#### Pull Requests
To submit a PR you can fork the repository (this basically makes your own linked copy at BGerwe/impedance.py) using the button at the top right of the ECSHackWeek/impedance.py page

[image]

Then clone your fork to your computer (`git clone https://github.com/BGerwe/impedance.py`), copy the changes over that you made, add, and commit (you probably want to check the changes locally -- see below -- first) to your own fork. Then, when you go back to your repo on GitHub you can click on the "New Pull Request" button to submit a pull request.

[image]

#### Checking the changes to docs locally
You can build the docs locally with make in the docs directory:
```bash
cd docs
make html
```
and then open `build/html/index.html` to see them!

Let me know if you run into any problems along the way!

BGerwe, 2019-10-09:
This is tremendously helpful, thanks!

commit b942f85f (2019-10-09): Fixes Gerischer element equation and documentation (#38)

Fixes #37
closed 2019-10-09 by mdmurbach