ECSHackWeek/impedance.py #66: validation.linKK manual mode doesn't work[BUG]

BGerwe, 2020-01-10:
Setting `c=None` and setting a manual number of RC elements e.g. `max_M=3` gives the error
```
fitLinKK() missing 1 required positional argument: 'Z'
```

Code to reproduce:
```
import numpy as np

# read data
data = np.genfromtxt('data/exampleData.csv', delimiter=',')

f = data[:,0]
Z = data[:,1] + 1j*data[:,2]

mask = f < 1000
f = f[mask]
Z = Z[mask]
```
```
from impedance.validation import linKK

M, mu, Z_linKK, res_real, res_imag = linKK(f, Z, c=None, max_M=3)
```

This should be a simple fix by changing line 94 from `p_values, mu = fitLinKK(f, M, Z)` to `p_values, mu = fitLinKK(f, ts, M, Z)`

mdmurbach, 2020-01-12:
Fixed by #67

closed 2020-01-12 by mdmurbach

mdmurbach, 2020-01-13:
Released in [v0.5.1](https://pypi.org/project/impedance/0.5.1/)