ECSHackWeek/impedance.py #83: Consistent naming scheme for circuit elements

mdmurbach, 2020-02-21:
Now that we can have multi-letter circuit elements, the options for how to name elements are _endless_... :joy: In reality, we should try to stick to a standard naming scheme to make it easier.

Currently, the elements that I think should be renamed (along with some early brainstorming...) are:
- A (Semi-infinite Warburg) -> W
- W (Blocked finite-length Warburg i.e. the one with the coth) -> Wb or Wo (looks like it's typically called open)
- [should add] (Transmissive finite-length Warburg, i.e. the one with the tanh) -> Wt or Ws (looks like it's typically called short)
- T (Macrohomogeneous porous electrode transmission line model) -> PE or TLMp?

BGerwe, 2020-03-13:
I'm inclined toward the Wo and Ws versions. I also have plans to include a finite-length Gerischer develope in [this paper for CO2 electrolysis](https://www.sciencedirect.com/science/article/pii/S0167273808003573#app1) since it's the model Jon and his undergrad are using. That element would be Gs since it's a Gerischer with a tanh term. 

Maybe this is better in a new issue, but I discovered a slightly annoying behavior with the new element naming method. Previously, elements could be followed by an underscore when writing the circuit string, but now it throws an error.

For example, using this circuit string `circ = 'R_0-p(R_1,C_1)-G_1'` raises a `ValueError`. I imagine this comes from how the `raw_element` variable is built and checked. 

```
for elem in extracted_elements:
    raw_element = ''.join(char for char in elem if char not in ints)
    num_params = check_and_eval(raw_element).num_params
    length += num_params
```

@nealde Do you think it's worth changing this to allow for `_` after the circuit element label? Or should we just discourage that practice?

mdmurbach, 2020-03-13:
Awesome. I'll rename the Warburg elements in my branch

I think we could just have the ignored characters (currently `ints`) be `0123456789_` and that should do the trick (means we can't ever name any elements with an underscore (i.e. `W_s`), but I think that's an okay tradeoff).

mdmurbach, 2020-03-14:
Elements like `R_0` work again as of 6c4b36e

BGerwe, 2020-03-16:
Are we ready to close this one too?

mdmurbach, 2020-03-16:
Yeah, I think so!

closed 2020-03-16 by BGerwe