ECSHackWeek/impedance.py #96: Improve test coverage

BGerwe, 2020-03-16:
Our latest coverage report shows:

```
Name                                    Stmts   Miss  Cover
-----------------------------------------------------------
impedance\__init__.py                       0      0   100%
impedance\models\__init__.py                0      0   100%
impedance\models\circuits\__init__.py       1      0   100%
impedance\models\circuits\circuits.py     183     15    92%
impedance\models\circuits\elements.py      81      1    99%
impedance\models\circuits\fitting.py      126      3    98%
impedance\preprocessing.py                154     21    86%
impedance\validation.py                    58     30    48%
impedance\visualization.py                 89      0   100%
-----------------------------------------------------------
TOTAL                                     692     70    90%
----------------------------------------------------------------------
```

I'll be tackling coverage in `validation` before I close out #69. Tt would be great to capture the loose ends in `circuits`, `elements` and `fitting`, and improve `preprocessing`. Can we make it to 100% before v1.0?

mdmurbach, 2020-03-16:
Yeah, I definitely think it's doable and a good goal! @dacb would be proud :) 

I'll start by setting up pytest so that we aren't including the tests/ directory (should be an easy fix that I've been meaning to do for a while now) and then I can start with the circuits module

mdmurbach, 2020-03-16:
PR for fixing the coverage calculations is #97

BGerwe, 2020-04-24:
Latest coverage report shows:
```
Name                                    Stmts   Miss  Cover
-----------------------------------------------------------
impedance/__init__.py                       0      0   100%
impedance/models/__init__.py                0      0   100%
impedance/models/circuits/__init__.py       1      0   100%
impedance/models/circuits/circuits.py     193      0   100%
impedance/models/circuits/elements.py      86      0   100%
impedance/models/circuits/fitting.py      126      0   100%
impedance/preprocessing.py                160     10    94%
impedance/validation.py                    94      0   100%
impedance/visualization.py                 89      0   100%
-----------------------------------------------------------
TOTAL                                     749     10    99%
```

Those missing 10 lines are from `readAutolab`. When we get our hands on an example data file we can get to 100% coverage!!

mdmurbach, 2020-04-24:
Awesome! 👏 👏 

To be fair, I don’t think that readAutolab function is from an actual autolab file. I think it’s from a custom file type that I created for the NLEIS experiments which is basically just a .csv — at first glance readAutolab and readCSV do the exact same thing. So while I think it’d be great to have a readAutolab function built off of a real autolab file, I think it makes sense to just get rid of the function...

BGerwe, 2020-04-24:
That's fair, but if you used AutoLab for NLEIS, did you never run regular EIS with it?

mdmurbach, 2020-04-24:
We always collected the time domain data (which is why there's a custom file type that was generated by the experimental protocol), so I don't know what the "normal" EIS file from Autolab would look like. Seems a little crazy, but I don't think I ever just ran EIS using their default settings (and saved the file)...

BGerwe, 2020-04-24:
@vwhu said he'll check for any files later today, but if he can't find anything I'm fine with removing the function and scrubbing it from our docs and JOSS paper draft.

BGerwe, 2020-04-24:
@vwhu is a hero and came through with the data file! This coverage report is a beautiful sight!
```
Name                                    Stmts   Miss  Cover
-----------------------------------------------------------
impedance/__init__.py                       0      0   100%
impedance/models/__init__.py                0      0   100%
impedance/models/circuits/__init__.py       1      0   100%
impedance/models/circuits/circuits.py     193      0   100%
impedance/models/circuits/elements.py      86      0   100%
impedance/models/circuits/fitting.py      126      0   100%
impedance/preprocessing.py                163      0   100%
impedance/validation.py                    94      0   100%
impedance/visualization.py                 89      0   100%
-----------------------------------------------------------
TOTAL                                     752      0   100%
```

closed 2020-04-25 by BGerwe