ECSHackWeek/impedance.py #99: [BUG] Defining 'Ws' or 'Gs' in custom circuits fails

BGerwe, 2020-03-16:
**Describe the bug**
Trying to define a custom circuit with 'Ws' or 'Gs' raises an error that the number of initial guesses doesn't match the circuit length. This seems to come from `extract_circuit_elements` incorrectly removing the `s` at the end of each element name.

**To Reproduce**
```
import sys
sys.path.append('../../../')

from impedance.models.circuits import CustomCircuit
```
```
circ = 'R_0-Gs_1-Ws_1'
inits =[10, 100, .1, .3, 50, 1]
sim = CustomCircuit(circuit=circ, initial_guess=inits)

>>AssertionError: The number of initial guesses + the number of constants (6) needs to be equal to the circuit length (4)
```

**Expected behavior**
The circuit should be successfully created with 6 initial guess parameters. The number of parameters for each element are: R_0 = 1, Gs_1 = 3, Ws_1 = 2.

**Additional context**
Running the following shows that `Gs` and `Ws` are incorrectly identified as `G` and `W`
```
from impedance.models.circuits.fitting import extract_circuit_elements
extract_circuit_elements(circ)

>>['R_0', 'G_1', 'W_1']
```

BGerwe, 2020-03-16:
The easiest solution seems to be renaming `Gs` and `Ws`, but the issue really stems from how p_string is constructed in:
```
def extract_circuit_elements(circuit):
    extracted_elements = []
    p_string = [x for x in circuit if x not in 'ps(),-/']
```

mdmurbach, 2020-03-16:
Ah, so it's because we've used `s` to dictate series...

I think there's an elegant way to do this with regular expressions rather than just ignoring characters, but I haven't played around with it much.

mdmurbach, 2020-03-16:
@BGerwe I think we should just be able to remove `s` and `/` (since we don't us it anymore) from the line you pasted above?

BGerwe, 2020-03-16:
That seems to work! Something else I noticed during this, the way elements with >1 parameters is a little confusing now. Ex:
```
Initial guesses:
    R_0 = 1.00e+01 [Ohm]
  G_1_0 = 1.00e+02 [Ohm]
  G_1_1 = 1.00e-01 [sec]
    W_1 = 5.00e+01 [Ohm sec^-1/2]
```

But we could have it print as
```
Initial guesses:
    R_0 = 1.00e+01 [Ohm]
  G_1_p[0] = 1.00e+02 [Ohm]
  G_1_p[1] = 1.00e-01 [sec]
    W_1 = 5.00e+01 [Ohm sec^-1/2]
```
without changing much. Are you fine with including that in this issue/PR?

mdmurbach, 2020-03-17:
Yeah, I think the double underline is a bit ugly haha. Maybe it’d be nice to have it be something like latex (I.e. Gs_{1,1}).

This does affect a few things though so probably best for a new PR.

PR #101 (2020-03-17): remove check for s and / in extract_circuit_elements
Fixes #99
closed 2020-03-17 by BGerwe