ECSHackWeek/impedance.py #108: [Element] support for modified inductance element

hkennyv, 2020-04-25:
Hey! I'd like to request adding a modified inductance element to `impedance.models.circuits.elements`.

Some background info on this element:
https://www.biologic.net/documents/battery-eis-modified-inductance-element-electrochemsitry-application-note-42/

The equation is similar to the CPE, except with inductance instead of capacitance (apologies, don't know how to render mathjax here):

Z = L(jw)^alpha

I have a fork with this component implemented as `M` (shown below), but it looks like there's support for multi-letter elements now so maybe we can find a more suitable representation?

```python
def M(p, f):
    """ defines a modified inductor
    .. math::
        Z = (L \\times j 2 \\pi f)^\\alpha}
     where :math:`L` = p[0] and :math:`\\alpha` = p[1].
     """

    typeChecker(p, f, M.__name__, 2)
    omega = 2*np.pi*np.array(f)
    inductance = p[0]
    alpha = p[1]

    return (inductance*1j*omega)**alpha
```

I wouldn't mind putting in the PR myself and adding some tests. How's this sound? Is there any other demand for this?

BGerwe, 2020-04-25:
Adding this element seems fine to me, I've never heard of this element being used but I don't see any harm in adding it to our list. Before submitting the PR please add a citation to the doc string for the academic article used in the application note, if there is one. I think the element name should either follow what's in the application note (`La`), or `Lf` to follow the naming convention used by Bernard Boukamp for fractal elements (which I'm guessing is part of the derivation for this element). Other than that, just look at how other elements are implemented and follow those conventions. 

And, yeah, please add tests for the element, we just hit 100% coverage yesterday and we're hoping to keep it there 😃

hkennyv, 2020-04-25:
wow gratz on 100% cov! i can work on this one and get a PR ready for review in the next couple days :)

dt-schwartz, 2020-04-28:
The biologic application note has a couple of publications cited at the end. I have not looked at them, but if they have the correct form of modified inductance and show the high & low frequency behaviors, you may want to cite those since they have DOIs and are long-lived archived docs.

closed 2020-04-28 by hkennyv