ECSHackWeek/impedance.py #111: [BUG] stale documentation for getting started

hkennyv, 2020-04-30:
**Describe the bug**
Documentation is stale for `getting-started.rst` and `getting-started.ipynb`, specifically where it still uses `impedance.circuits` and `impedance.plotting`.

**To Reproduce**
See [`getting-started.rst`](https://github.com/ECSHackWeek/impedance.py/blob/master/docs/source/getting-started.rst)

**Expected behavior**
Use the new API:

```python
from impedance.models.circuits import CustomCircuit
```

and

```python
from impedance.visualization import plot_nyquist
```

**Additional context**
i've started a PR to fix this, need to update the notebook still

BGerwe, 2020-04-30:
Good catch! I think all of the example notebooks were updated for the new module structure, but this was overlooked.

mdmurbach, 2020-05-01:
Yeah, great catch. Thanks for already starting a PR as well! 😁

PR #112 (2020-05-03): update getting-started docs & example
* [x] fix imports `getting-started.rst`
* [x] update circuit elements in `getting-started.rst`
* [x] update `getting-started.ipynb`

**updating data imports**
* [x] getting-started
* [x] fitting-example
* [x] model_io_example
* [x] validation_example
* [x] plotting_example

closes #111
closed 2020-05-03 by BGerwe