ECSHackWeek/impedance.py #121: Errors in Example Notebooks

EricaEgg, 2020-07-17:
Hi impedance.py team,

I found some errors while using some of your example notebooks. See below:

1) In the notebook, fitting_example.ipynb, part 4a, numpy is called without an import. Import line needs to be added in this block or earlier.
2) In the notebook plotting_example.ipynb, part 3a, I am getting an error. There could be a bug in the function circuit.plot. (error attached as photo)
[image]

Let me know if I can clarify anything!

mdmurbach, 2020-07-18:
Thanks for reporting this! I think this is an error with the javascript for vega (from altair) being loaded. Do you have an ad blocker running per chance?

BGerwe, 2020-07-20:
For what it's worth, I just ran all the example notebooks after [Matt's fixes](https://github.com/ECSHackWeek/impedance.py/commit/2797d5fd68275855795cfb6e7d5610e77bea2314) in a fresh virtual environment and didn't get any errors.

EricaEgg, 2020-07-20:
Thanks @mdmurbach and @BGerwe! I just re-ran it, and it is working perfectly.

BGerwe, 2020-07-21:
Thanks for your work with the review @EricaEgg!

closed 2020-07-22 by mdmurbach