ECSHackWeek/impedance.py #127: Error Analysis

Gold247, 2020-08-26:
Hello all,
First i want to appreciate y'all for this wonderful package.
I'd like to ask if its possible to further develop the error analysis aside the standard error to include the following;
chi-square, reduced chi-square, Akaike info criteria, Bayesian info criterion and the correlation matrix between the fitted parameters. 

Best Regards

BGerwe, 2020-08-28:
Hi @Gold247, sorry we didn't get back to you quickly. Your original post about including other error analysis metrics seems valuable to me. Feel free to reopen this issue if you see fit.

Gold247, 2020-08-29:
HI @BGerwe ,
thanks for replying.
I am new to python, however i discovered the python module LMFIT has better error analysis.
I think if there is way to use the leastsq of lmfit, by passing the residuals of the circuit function to leastsq, it will be amazing. please pardon me if what i'm saying is wrong as my knowledge of python is limited.
Best regards 

[image]

mdmurbach, 2020-09-15:
Just wanted to re-add the original thoughts on error analysis:

> Hello all,
> First i want to appreciate y'all for this wonderful package.
> I'd like to ask if its possible to further develop the error analysis aside the standard error to include the following; [...]

I think adding additional outputs to the fitting results could be a great addition to the project (I think one big opportunity we're still missing that we've always talked about is a good way to do model selection or comparing across models/fits). It's also worth thinking about changing the fitting routine to something like [lmfit](https://github.com/lmfit/lmfit-py) if it makes sense...

radmaker, 2021-02-08:
In one of the examples the error of the fit is calculated with the absolute value in the denominator:

res_meas_real = (Z - meas_model.predict(f)).real/np.abs(Z)
res_meas_imag = (Z - meas_model.predict(f)).imag/np.abs(Z)

Would't it make more sense to calculate the complex errors with respect to the real and imaginary part in the denominator as follows:

res_meas_real = (Z - meas_model.predict(f)).real/np.real(Z)
res_meas_imag = (Z - meas_model.predict(f)).imag/np.imag(Z)