ECSHackWeek/impedance.py #128: [BUG] Prediction fails on circuits of one element

lktsui, 2020-08-28:
**Describe the bug**
If the circuit only has one element (a single resistor or capacitor) in it, the prediction fails with a UnboundLocalError

**To Reproduce**
```
from impedance.models.circuits import CustomCircuit
import numpy as np
test_circuit = CustomCircuit(circuit='C_1', initial_guess = [1.0,])
print(test_circuit)
frequency_range = np.linspace(0,100)
test_circuit.predict(frequency_range)
```

```
Circuit string: C_1
Fit: False

Initial guesses:
    C_1 = 1.00e+00 [F]

Simulating circuit based on initial parameters
---------------------------------------------------------------------------
UnboundLocalError                         Traceback (most recent call last)
<ipython-input-9-0336b0f9ba3e> in <module>
      4 print(test_circuit)
      5 frequency_range = np.linspace(0,100)
----> 6 test_circuit.predict(frequency_range)

~/Documents/Programs/impedance.py/impedance/models/circuits/circuits.py in predict(self, frequencies, use_initial)
    156                                      *self.initial_guess,
    157                                      constants=self.constants, eval_string='',
--> 158                                      index=0)[0],
    159                         circuit_elements)
    160 

~/Documents/Programs/impedance.py/impedance/models/circuits/fitting.py in buildCircuit(circuit, frequencies, constants, eval_string, index, *parameters)
    204         split = parallel
    205 
--> 206     for i, elem in enumerate(split):
    207         if ',' in elem or '-' in elem:
    208             eval_string, index = buildCircuit(elem, frequencies,

UnboundLocalError: local variable 'split' referenced before assignment
```

**Expected behavior**
It should predict the impedance response of a single resistor / capacitor.

**Additional context**
I was sanity checking a circuit model by testing a single resistor and it failed with this error.

lktsui, 2020-08-28:
Test code for a single resistor.

```
from impedance.models.circuits import CustomCircuit
import numpy as np
test_circuit = CustomCircuit(circuit='R_1', initial_guess = [1.0,])
print(test_circuit)
frequency_range = np.linspace(0,100)
test_circuit.predict(frequency_range)
```

```
Circuit string: R_1
Fit: False

Initial guesses:
    R_1 = 1.00e+00 [Ohm]

Simulating circuit based on initial parameters
---------------------------------------------------------------------------
UnboundLocalError                         Traceback (most recent call last)
<ipython-input-10-cbc428542153> in <module>
      4 print(test_circuit)
      5 frequency_range = np.linspace(0,100)
----> 6 test_circuit.predict(frequency_range)

~/Documents/Programs/impedance.py/impedance/models/circuits/circuits.py in predict(self, frequencies, use_initial)
    156                                      *self.initial_guess,
    157                                      constants=self.constants, eval_string='',
--> 158                                      index=0)[0],
    159                         circuit_elements)
    160 

~/Documents/Programs/impedance.py/impedance/models/circuits/fitting.py in buildCircuit(circuit, frequencies, constants, eval_string, index, *parameters)
    204         split = parallel
    205 
--> 206     for i, elem in enumerate(split):
    207         if ',' in elem or '-' in elem:
    208             eval_string, index = buildCircuit(elem, frequencies,

UnboundLocalError: local variable 'split' referenced before assignment
```

BGerwe, 2020-08-29:
I can replicate this issue locally and it doesn't seem to matter what element is input. I think the issue is we don't have logic for a scenario where neither series nor parallel markers are in the string:
```
parallel = parse_circuit(circuit, parallel=True)
series = parse_circuit(circuit, series=True)

if series is not None and len(series) > 1:
    eval_string += "s(["
    split = series
elif parallel is not None and len(parallel) > 1:
    eval_string += "p(["
    split = parallel

for i, elem in enumerate(split):
```

closed 2020-08-31 by mdmurbach