ECSHackWeek/impedance.py #130: [Ls - de Levie Pore _Finite Length Element]

ravuru66, 2020-09-14:
This element describes the impedance of a pore with conductive and reactive pore walls. This is in contrast to Finite Length Warburg element where the pores are non-reactive (infinite polarization resistance).

Z(w) = (R/(gamma)^1/2) * coth((gamma)^1/2)

where Z is the impedance , R is the resistance, w is the angular frequency and 
gamma = (1/A)+B*(jw)^phi 

The following referenes give a detailed description of the electrochemical  model:

1. A.Lasia, "Impedance of porous electrodes", Modern Aspects of electrochemistry, "Modellingand Numerical Simulations," vol.43, p.67-138, M.Schlesinger, Ed.,Springer, 2009, IsBN: 978-0-387-49580-4.

2. r De Levie, Adv. electrochem. Elecrochem. eng., 6(1967) 329.

mdmurbach, 2020-09-14:
Hi @ravuru66,

Thanks for reaching out and for including the equation and citations! 🚀 

Would you want to take a crack at adding the new element? I (and others, I'm sure) would be happy to help review or answer any questions you have. A good starting point for how to get started making changes locally can be found in [CONTRIBUTING.md](https://github.com/ECSHackWeek/impedance.py/blob/master/CONTRIBUTING.md#repository-setup) and the other elements should provide a good starting point for adding a new one. If not, we can add the "help wanted" tag and get it added sometime soon.

Thanks again!

ravuru66, 2020-09-15:
Hi @mdmurbach ,

Thank you for your immediate response.

Yes, I would like to take a crack at adding the new element because I am going to work on this impedanec.py package in the coming months and would like to contribute to this package by elements from my research work. So it would be nice for me to add element on my own if you guide me how to do so.

Best Regards

BGerwe, 2023-06-24:
With the addition of custom elements in #219, can we close this?