ECSHackWeek/impedance.py #133: Weighted fitting based on |Z|

stephendkang, 2020-09-16:
Currently the fitting is based on real and img values of Z without any weighting. This approach puts a bias on low frequency data points. I believe the standard weighting scheme used in many impedance fitting software is to use the absolute impedance.

This can be easily implemented. In the `circuit_fit` function where you call `curve_fit` from scipy, you can do something like:
`abs_Z = np.abs(Z)
sigma = np.hstack([abs_Z, abs_Z])`

which will be the inverse weights. The `curve_fit` in scipy has an optional `sigma` argument that can be used for weighting, to which `sigma` can be supplied.

This small change greatly improves the EIS fitting quality in some data sets.

Not sure whether it's better to add this as an option or to change the default fitting scheme. So I'm submitting this as a feature request.

mdmurbach, 2020-09-28:
Hi @stephendkang,

Thanks, that's a great idea. A lot of times it is useful to pass one over the experimental variance as the weighting factor for the fit as well. I just opened a PR that exposes all the kwargs for scipy.optimize.curve_fit, so you should be able to easily pass sigma now! It would be awesome if you could share how you go about weighting the fit once you've got an example so we can add a code example to the FAQ or fitting example docs.

BGerwe, 2020-10-01:
Addressed by #136

closed 2020-10-01 by BGerwe