ECSHackWeek/impedance.py #138: Bug in creating custom circuits - UnboundLocalError

richinex, 2020-10-01:
I tried to create a  modified randle's circuit (circ_string = 'R_0-p(CPE_1,(R_1-p(R_2,W_1)))'   )by using a custom string but I get an error:

UnboundLocalError: local variable 'split' referenced before assignment

The circuit is a modified Randle's ciruit with the planar infinite Warburg in parallel with a resistor.

Kindly assist thank you.

BGerwe, 2020-10-01:
Hello, 

It seems that circuit string is missing a 'p' at the second parentheses. Try `circ_string = 'R_0-p(CPE_1,p(R_1-p(R_2,W_1)))'`

richinex, 2020-10-02:
> Hello,
> 
> It seems that circuit string is missing a 'p' at the second parentheses. Try `circ_string = 'R_0-p(CPE_1,p(R_1-p(R_2,W_1)))'`

still the same error when I try your suggestion: 

 `from impedance.models.circuits import CustomCircuit

circuits = []

circ_string = 'R_0-p(CPE_1,p(R_1-p(R_2,W_1)))'
initial_guess = [1000,1,1, 1000,1000,1000]

bounds = ([1e-1, 1e-1, 1e-15, 1e-2, 1e-2, 1e-2], [1e9, 1e0, 1e1, 1e15, 1e13, 1e15])

# Now loop through data list to create circuits and fit them
for f, Z, filename in zip(freqs, Zs, all_files):
    name = filename.split('_')[-1].split('.')[0]
    circuit = CustomCircuit(circ_string, initial_guess=initial_guess, name=name)
    circuit.fit(f, Z, bounds = bounds)
    circuits.append(circuit)`

ERROR: 
C:\ProgramData\Anaconda3\envs\impedance-analysis\lib\site-packages\impedance\models\circuits\fitting.py in buildCircuit(circuit, frequencies, constants, eval_string, index, *parameters)
    204         split = parallel
    205 
--> 206     for i, elem in enumerate(split):
    207         if ',' in elem or '-' in elem:
    208             eval_string, index = buildCircuit(elem, frequencies,

UnboundLocalError: local variable 'split' referenced before assignment

hkennyv, 2020-10-06:
@richinex it looks like you're missing a term in your second parallel group:

`p(R_1-p(R_2,W_1))` only has 1 item in parallel. did you mean to put `R_1` in parallel with `p(R_2,W_1)`?

otherwise, you need to put something else in parallel with the `R_1-p(R_2,W_1)` term

closed 2020-12-06 by mdmurbach