ECSHackWeek/impedance.py #141: Add method for saving processed file

aokomorowski, 2020-11-13:
The library has a plenty of functions that allow processing data into single f, Z series. I think it would be nice to have an easy option to save these series into a file. I will create PR with solution shortly.

mdmurbach, 2020-11-17:
Hi @aokomorowski. Sounds good, thanks. I think it might make sense to have a simple save data function in preprocessing.py. You could basically just use pandas.to_csv() and make it compatible with the read_csv() function already in there https://github.com/ECSHackWeek/impedance.py/blob/cf63337fa800ce768373163701bd175e74913f0e/impedance/preprocessing.py#L393-L399

BGerwe, 2021-12-02:
Closing this because it was addressed by #167

closed 2021-12-02 by BGerwe