ECSHackWeek/impedance.py #144: Error when trying to fit , keeping one of the parameter keeping constant in an element

ravuru66, 2020-11-29:
when there are more than one parameter in a element, for example in short warburg element there are two parameters Z0, tau. when I am trying to fit keep one of the parameters constant. it is giving an error.

my code

from impedance import preprocessing
from preprocessing import importdata
from impedance.models.circuits import CustomCircuit
import matplotlib.pyplot as plt
from impedance.visualization import plot_nyquist

frequencies, Z = importdata("EIS_20-05-14_04.28.39_ser04.txt")
   
# keep only the impedance data in the first quandrant
frequencies, Z = preprocessing.ignoreBelowX(frequencies, Z)
      
circuit = 'R0-Ws0'
initial_guess =[0.1, 0.3]

**circuit = CustomCircuit(circuit, initial_guess=initial_guess, constants={"Z0":0.2})**
circuit.fit(frequencies, Z)

s_value = circuit.parameters_
Z_sim = circuit.predict(frequencies, use_initial=True)
Z_fit = circuit.predict(frequencies)
  
fig, ax = plt.subplots()
plot_nyquist(ax, Z, fmt='o')
plot_nyquist(ax, Z_fit, fmt='*')
plot_nyquist(ax, Z_sim, fmt='-')
plt.legend(['Data', 'Fit', 'simulation'])
plt.show()

error:
File "C:\ProgramData\Anaconda3\lib\site-packages\scipy\optimize\_lsq\least_squares.py", line 788, in least_squares
    raise ValueError("Inconsistent shapes between bounds and `x0`.")

ValueError: Inconsistent shapes between bounds and `x0`.

Is it possible to fit the circuit keeping of the parameter constant in an elemen with impedance software? If so how to do that.

please let me know your answer

mdmurbach, 2020-12-06:
Hi @ravuru66, it looks like you may just have the element defined wrong in the constants dictionary (should be the name of the element in the circuit string to hold constant, i.e. `R0` instead of `Z0`). Please let us know if there are still issues after this change? Thanks!

mdmurbach, 2021-01-26:
Feel free to reopen if the above didn't fix the issue, thanks!

closed 2021-01-26 by mdmurbach