ECSHackWeek/impedance.py #147: Fittinf accuracy

yuchengluo, 2021-01-21:
Hi, thanks for the nice code first of all. I met some problems while using the code and need your help.
I am doing the fitting with several groups of data (data at different SOC level) from one single battery, so I give them the same initial-guess-value at the beginning. As you can see in the pictures, some of the fitting results are good, but others are not so precise, especially when it comes to high/low SOC level dataset. I have also added bounds for the data but I dont't think it works.
[image]
[image]

circuit = 'R0-p(R1,C1)-p(R2-Wo1,C2)'
initial_guess = [.8, 1, 20, .1, 1, 100, .02]
bounds = ([0.7, 0.6, 20, 0.06, 0.6, 75, 0.01],[2, 6, 60, 0.4, 15, 290, 0.1])

Do you have any experience or ideas about how to fix the problem?

Thanks for your time and patience!

mdmurbach, 2021-01-26:
If you know that the curves you are fitting vary smoothly over SoC, you can try passing the fit parameters from the previous SoC fit as the initial_guess values for the current SoC fit.

yuchengluo, 2021-02-01:
> If you know that the curves you are fitting vary smoothly over SoC, you can try passing the fit parameters from the previous SoC fit as the initial_guess values for the current SoC fit.

Hi mdmurbach, thanks for your help. I tried to do the same way as you mentioned, use the previous fitting parameter to fit the inaccurate dataset, but the result was still not ideal, the curves look basically the same.

closed 2021-02-21 by yuchengluo