ECSHackWeek/impedance.py #149: Travis CI shutting down

mdmurbach, 2021-01-26:
Travis-CI.org is shutting down and there appears to be [some issues](https://news.ycombinator.com/item?id=25338983) with open source projects getting credits with Travis-CI.com. 

We should figure out how to switch the testing over to GitHub Actions (or another CI platform) over the next few weeks 🙂

ml-evs, 2021-07-29:
Hi there, I'm interested in using this package as part of a broader echem/materials data management suite - do you want a hand migrating to GitHub actions? I have a fair bit of experience with it.

mdmurbach, 2021-09-09:
That's great. Sorry for the super long delay. Really appreciate the PR!

closed 2021-09-09 by mdmurbach

ml-evs, 2021-09-09:
No problem, hope it's helpful!

I've just noticed that the CI documentation under CONTRIBUTING.md is now out of date, but should be fixable with `s/Travis/GitHub Actions/g`.

PR #171 (2021-09-09): Migrate CI from Travis to GitHub Actions
This PR is mostly just a straight swap from Travis to GitHub Actions (closes #149). The only addition is the testing of multiple Python versions (3.6, 3.7, 3.8 and 3.9) in parallel.