ECSHackWeek/impedance.py #152: Wierd values of R1, R2, C1, C2 in the example and also my fitting result

yuchengluo, 2021-02-02:
I found some wierd situtations in my fitting and I can also find it in the example.

Firstly, my fitting:

Circuit string: R0-p(R1,C1)-p(R2-W1,C2)
Fit: True

Initial guesses:
     R0 = 1.04e+00 [Ohm]
     R1 = 1.90e+00 [Ohm]
     C1 = 2.40e+01 [F]
     R2 = 1.10e-01 [Ohm]
     W1 = 1.10e-01 [Ohm sec^-1/2]
     C2 = 1.50e-02 [F]

Fit parameters:
     R0 = 1.51e+00  (+/- 2.31e-03) [Ohm]
     R1 = 1.35e+00  (+/- 1.11e-02) [Ohm]
     C1 = 3.08e+01  (+/- 2.90e-01) [F]
     R2 = 1.01e-01  (+/- 2.34e-03) [Ohm]
     W1 = 7.71e-02  (+/- 7.76e-04) [Ohm sec^-1/2]
     C2 = 3.09e-02  (+/- 1.77e-03) [F]
[image]

The fitting result looks good. The thing I don't understand is:
Why the values of R1 and C1 are larger than R2 and C2? 
Because 1/(RC) should be the position of the semicircle's peak, so normally if there are two RC, then1/(R1C1) would be on the left side (higher frequency), so 1/(R1C1) is bigger than 1/(R2C2), which means the values of R1 and C1 should be smaller than R2 and C2.

This can be proved by other literatures:
[image]

As you can see, Rsei represents the first semi-circle and Rct is the second one.

And I also found in the examples the 1/(R1C1) is bigger, which doesn't make sense.
[image]

Could you please tell be the reason? Thank you.

BGerwe, 2021-02-02:
Hi @yuchengluo,

I think what you're seeing is a consequence of how we interpret a model vs how it's represented mathematically. What I mean by that is you've constructed a model attributing certain elements to EIS features a priori. However, the curve fitting function just wants to minimize the sum of residuals and doesn't care as much where the elements are "supposed" to go. 

In your case `p(R1,C1)` and `p(R2-W1,C2)` obviously aren't the same, but the fit here associates `W1` with the high-frequency arc. I think the depressed semi-circular shape of the high-frequency arc is partially responsible. Your model uses capacitors, whereas the attached figure shows a model with CPEs. Replacing your `C` with `CPE` may result in a more sensible fit. Beyond that, try adding bounds to the fit to "force" `p(R1,C1)` to fit the high-frequency arc.

I hope that helps :smile:

yuchengluo, 2021-02-03:
Hi @BGerwe,

Thanks for your answer! I just replaced the capacitors with CPEs, and got actually better results than before. Thank you for your advices!

And addtionally I then tried to set the bounds to force p(R1,C1) to fit the high frequency and also force R2 and C2 to a bigger range. The results were however not so good as the one without the bounds.
So if I want the perfect results, which means I don't use any bounds. Just like you said, let the fitting function minimize the sum of residuals, then how can I explain the situation, that the R1 and C1 are bigger than R2 and C2? Because the circuit 'R0-p(R1,C1)-p(R2-W1,C2)' is also used in other literatures, and how can their parameters represent the curve respectively (R1C1->first semi-circle, R2C2-> second one, W->diffusion tail)?

Thank you again for your kindly help!

BGerwe, 2021-02-12:
Hi @yuchengluo,

Perhaps it's best to only bound `p(R1,C1)` and leave `p(R2-W1,C2)` unbounded. I've also noticed that CPEs are especially sensitive to the initial guess of α and Q—so you may want to try varying those to get a better fit. 

Finally, you can sometimes refine your fit by passing the parameters as the initial guess and refitting.

For example:
```
circ_str = 'R0-p(R1,CPE1)'
inits = [20, 60, 1e-1, .1]

circ = CustomCircuit(circuit=circ_str, initial_guess=inits)

circ.fit(f, Z)
circ.initial_guess = circ.parameters_
circ.fit(f, Z)
```

BGerwe, 2021-02-21:
Hi @yuchengluo,

Just checking in to see if you're still having trouble with this. Also, it seems very related to #147—do you mind if I close that issue and continue any further discussions here?

yuchengluo, 2021-02-21:
Hi @BGerwe,

sorry for the late reply! Thanks a lot, the problem has been solved, thank you for your great tips, the results get much better!

closed 2021-02-21 by yuchengluo

BGerwe, 2021-02-21:
Hi @yuchengluo,

I'm glad to hear it! Thank you for using this package and starting these discussions.

2410337637, 2022-10-26:
Hi there!

I would like to know how to bound only p(R1,C1) and leave p(R2-W1,C2) unbounded? Can you show me the relevant code?

Thank you!