ECSHackWeek/impedance.py #156: Poor fitting - why the RC semicircle seems off?

CDavis222, 2021-03-05:
Hello,
I am wondering why my fit is having this issue- the semicircle RC element seems to be about the right dimensions, just is oddly offset. As a new user, is there something I'm missing to get the fit better?

circuit elements and initial guess:
circ_string_1 = 'R0-p(R1-W1,C1)'
initial_guess = [0.01, 100, 40000, 1e-6]

Thanks for your help!

[image]

svmchop, 2021-04-21:
First of all, use bounds. Your 'R0' is clearly much less than your initial value. Giving wider bounds will help in improving your fitting.
Second, I think you should use one more voigt element to fit the data.. The original nyquist is clearly showing some kind of depressed semicircle in mid-frequency region.

krarmd, 2021-08-03:
Can you give an example of how to set low and upper bounds for a specific circuit element? I see the option for setting a constant value, but it's not clear to me how to specify a range of values for a CPE or Warburg element.

"
def circuit_fit(frequencies, impedances, circuit, initial_guess, constants={},
                bounds=None, global_opt=False, **kwargs)

    bounds : 2-tuple of array_like, optional
        Lower and upper bounds on parameters. Defaults to bounds on all
        parameters of 0 and np.inf, except the CPE alpha
        which has an upper bound of 1
"

mdmurbach, 2021-09-10:
Thanks for updating with your working code, krarmd. CDavis222, feel free to reopen if you're still having issues!

closed 2021-09-10 by mdmurbach