ECSHackWeek/impedance.py #164: Applying weighting to circuit.fit()

FTHuld, 2021-04-07:
I have a CustomCircuit() that I am trying to apply a weighting to the fitting function ([e.g. the one used by the Metrohm software - PDF warning](http://www.autolabj.com/appl.files/appl%20note2015/Autolab_Application_Note_EIS05.pdf)). Since the impedance fitting uses scipy.optimize.curve_fit() I assume that it should be possible to use the weighting functionality `sigma` from curve_fit() to do this.

Unfortunately, not matter what values I input for sigma (up to and including nonsense like strings), I get `ValueError: 'sigma' has incorrect shape.`

See example below (it's the one from the "getting started" section)

```
from impedance import preprocessing
from impedance.models.circuits import CustomCircuit

# Load data from the example EIS result
frequencies, Z = preprocessing.readCSV('./exampleData.csv')

# keep only the impedance data in the first quandrant
frequencies, Z = preprocessing.ignoreBelowX(frequencies, Z)
sigma = abs(Z)

circuit = 'R0-p(R1,C1)-p(R2-Wo1,C2)'
initial_guess = [.01, .01, 100, .01, .05, 100, 1]

circuit = CustomCircuit(circuit, initial_guess=initial_guess)
#circuit.fit(frequencies, Z, sigma=sigma)
```

Note that `np.shape(sigma)==np.shape(frequencies)==np.shape(Z)` is `True`.

Any ideas as to what might be going on?

Thanks,

F

BGerwe, 2021-04-07:
Hi Fredrik,

First off, thank you for using the package and opening this issue!

I haven't used the weighting a ton, but my guess is the shape problem arises from how we deliver data to `curve_fit`
https://github.com/ECSHackWeek/impedance.py/blob/463df381211197a74884c516e9e2800ce508a94d/impedance/models/circuits/fitting.py#L118-L120

As you can see, we separate the real and imaginary parts of the data, so the weight array you provide should be `2 * len(Z)`. Note that imaginary parts are concatenated to the end of the real parts:
```
re = np.linspace(0, 10, num=11)
im = np.linspace(20, 30, num=11)

np.hstack((re, im))

>>> array([ 0.,  1.,  2.,  3.,  4.,  5.,  6.,  7.,  8.,  9., 10., 20., 21.,
       22., 23., 24., 25., 26., 27., 28., 29., 30.])
```

Let me know if that works for you! 😄

FTHuld, 2021-04-14:
Ah that makes sense now.

Thank you very much for your help!

closed 2021-04-14 by FTHuld