ECSHackWeek/impedance.py #168: [BUG] Unable to use basinhopping algorithm

svmchop, 2021-04-21:
I wanted to use basinhopping algorithm to fit my data using circuit string: R0-p(CPE1,R1-Wo_1-p(CPE2,R2))..
But, I am being shown this error:
TypeError: circuit_fit() got an unexpected keyword argument 'global_opt'

Although its clearly defined in the docs that to use basinhopping, change global_opt, but, circuit_fit is unable to recognize the argument.

mdmurbach, 2021-04-29:
Can you confirm the version of impedance.py that you're using?

```
import impedance
print(impedance.__version__)
```

The basinhopping algorithm was added in v1.2. You can upgrade if necessary with `pip install impedance --upgrade`

mdmurbach, 2021-09-09:
Assuming this ended up being a version issue. Feel free to reopen if you're still having issues!

closed 2021-09-09 by mdmurbach