ECSHackWeek/impedance.py #169: Fix certain parameters before attempting fit

devangs3, 2021-04-22:
I am unable to fix one or multiple circuit parameters when it comes to trying a circuit model fit as Zview allows. Isthere a way the same can be done in this implementation? Here is a link for reference http://lacey.se/science/eis/fitting-with-real-data/

[image]

svmchop, 2021-04-23:
Hi @devangs3 
There is an option to make circuit parameters constant. 
[image]

devangs3, 2021-04-27:
Could this be added to the documentation page? Or if you can provide a link to this syntax? @svmchop

closed 2021-04-27 by devangs3