ECSHackWeek/impedance.py #170: [BUG] Fitting with multiple CPE gives errors

devangs3, 2021-05-24:
I am trying to replicate fitting of this custom circuit R_0 - p(CPE_1, R_1 - p(R_2,CPE_2) ) using the following statement

`customConstantCircuit = CustomCircuit(initial_guess=[100e3, 1.3e-12, .96, 1e9, 1e11, 6.6e-12, .9 ],
                                      CPE=True,
                                      constants={'CPE_1_1': 0.96, 'CPE_2_1': 0.9},
                                      circuit='R_0 - p(CPE_1, R_1 - p(R_2,CPE_2) )')`

But, I see the error 

TypeError : __init__() got an unexpected keyword argument 'CPE'

What am I doing wrong?

sodeve19, 2021-05-24:
Not a contributor, but I have been using the API. I don't think you need this: CPE=True. Also you are giving initial guesses to the constants, you should replace the values for None, on the initial guess. Likes this:

`customConstantCircuit = CustomCircuit(initial_guess=[100e3, None, None, 1e9, 1e11, 6.6e-12, .9 ], constants={'CPE_1_1': 0.96, 'CPE_2_1': 0.9}, circuit='R_0 - p(CPE_1, R_1 - p(R_2,CPE_2) )')`

Hope that helps.

devangs3, 2021-05-24:
> Not a contributor, but I have been using the API. I don't think you need this: CPE=True. Also you are giving initial guesses to the constants, you should replace the values for None, on the initial [...]

I get the error:
ValueError : CPE not in allowed elements (dict_keys(['Gs', 'CPE', 'TLMQ', 'W', 's', 'p', 'La', 'K', 'Ws', 'Wo', 'G', 'R', 'C', 'L', 'T', '__builtins__']))

sodeve19, 2021-05-24:
Okay, I missed you had CPE_1_1 and CPE_2_1 on the constants. I assume you want CPE_1 constant, and the values are for CPE and alpha. In that case this should work:

`customConstantCircuit = CustomCircuit(initial_guess=[100e3, None, None, 1e9, 1e11, 6.6e-12, .9 ], constants={'CPE_1_0': 0.96, 'CPE_1_1': 0.9}, circuit='R_0 - p(CPE_1, R_1 - p(R_2,CPE_2) )')`

devangs3, 2021-05-24:
> If what you want is to keep the alphas for both CPEs constant, this may work, but I have not tried it:
> 
> `customConstantCircuit = CustomCircuit(initial_guess=[100e3, 1.3e-12, None, 1e9, 1e11, 6.6e-12, None ], constants={'CPE_1_1': 0.96, 'CPE_2_1': 0.9}, circuit='R_0 - p(CPE_1, R_1 - p(R_2,CPE_2) )')`

Thanks, this is exactly what I want, but I am getting the same error as last:
ValueError : CPE not in allowed elements (dict_keys(['Gs', 'CPE', 'TLMQ', 'W', 's', 'p', 'La', 'K', 'Ws', 'Wo', 'G', 'R', 'C', 'L', 'T', 'builtins']))

sodeve19, 2021-05-24:
Yes, I am not sure if you can do that. Have you tried putting all the values for the CPE constant and check if that works?

`customConstantCircuit = CustomCircuit(initial_guess=[100e3, None, None, 1e9, 1e11, None, None ], constants={'CPE_1_0':1.3e-12, 'CPE_1_1': 0.96, 'CPE_2_0':6.6e-12, 'CPE_2_1': 0.9}, circuit='R_0 - p(CPE_1, R_1 - p(R_2,CPE_2) )')`

devangs3, 2021-05-24:
> Yes, I am not sure if you can do that. Have you tried putting all the values for the CPE constant and check if that works?
>
> `customConstantCircuit = CustomCircuit(initial_guess=[100e3, None, None, [...]

Tried this, no effect :(

ColdAdam, 2021-07-20:
@devangs3 
Just resolved a similar error, spaces in the circuit string are being carried into the elements throwing an error.
Try: circuit='R_0-p(CPE_1,R_1-p(R_2,CPE_2))'
and see if that resolved your issue.

devangs3, 2021-07-20:
> @devangs3 
> 
> Just resolved a similar error, spaces in the circuit string are being carried into the elements throwing an error.
> 
> Try: circuit='R_0-p(CPE_1,R_1-p(R_2,CPE_2))'
> 
> and see if that resolved your issue.

This worked!

closed 2021-07-20 by devangs3