ECSHackWeek/impedance.py #172: Generalize CSV preprocessing

xraymancs, 2021-08-11:
Would it be possible to add a read routine for the EZStat file format which has frequency, |Z|, phase, Re{Z} and -Im{Z} as its 5 columns with a header line.  The specific EZStat code would be as follows:

def readEZStat(filename):
    """ function for reading EZStat csv files
    Parameters
    ----------
    filename: string
        Filename of .csv file to extract impedance data from
    Returns
    -------
    frequencies : np.ndarray
        Array of frequencies
    impedance : np.ndarray of complex numbers
        Array of complex impedances
    """
    data = np.genfromtxt(filename, delimiter=',')

    f = data[1:, 0]
    Z = data[1:, 3] - 1j*data[1:, 4]

    return f, Z

hkennyv, 2021-08-21:
@xraymancs i think it would help if you could provide a couple sample data files? that way they can add some tests too!

mdmurbach, 2021-09-09:
Yep, agree with @hkennyv! An example file (even with fake data) would be really helpful here @xraymancs

xraymancs, 2021-09-09:
Sorry about the delay, Here it is.  I believe that this is the default but
it would be more fault tolerant if the input routine would use the column
headings to determine what to read in.  I cannot guarantee that the order
of the columns is always the same as what I get from my instrument.  I do
note that the subtraction that I showed in my code snippet is necessary asn
the file has already applied a negative to the data.

Carlo

mdmurbach, 2021-09-09:
I don't think the attachment worked if there was one? (At least I can't see in on the github website)

xraymancs, 2021-09-09:
I attached it to the email. Should I put it directly on github?

Carlo

xraymancs, 2021-09-09:
Here is the file I sent previously by email, sorry!
[cell_1.csv](https://github.com/ECSHackWeek/impedance.py/files/7133774/cell_1.csv)

xraymancs