ECSHackWeek/impedance.py #176: Fitting parameters uncertainty

emrebaburoglu, 2021-08-30:
I am currently using the package to fit a circuit to my nyquist plots and although it works, I can't seem to understand what this uncertainty (plus/minus) on the fit parameters mean or how they were calculated. I'd really appreciate it if someone could explain. I've attached an example of intial guesses and fit parameters to explain what I mean. 
Initial guesses:
R0 = 1.65e-02 [Ohm]
R1 = 8.77e-03 [Ohm]
C1 = 3.28e+00 [F]
R2 = 5.31e-03 [Ohm]
C2 = 2.32e-01 [F]
Wo1_0 = 6.37e-02 [Ohm]
Wo1_1 = 2.37e+02 [sec]
Fit parameters:
R0 = 3.22e-02 (+/- 2.31e-04) [Ohm]
R1 = 1.31e-02 (+/- 2.84e-04) [Ohm]
C1 = 2.19e+00 (+/- 1.24e-01) [F]
R2 = 7.96e-03 (+/- 3.10e-04) [Ohm]
C2 = 1.55e-01 (+/- 1.26e-02) [F]
Wo1_0 = 9.56e-02 (+/- 3.05e-03) [Ohm]
Wo1_1 = 2.38e+02 (+/- 1.73e+01) [sec]

Thank you,
Emre Baburoglu

petermattia, 2021-08-31:
Hi, the errors are standard deviations. For the standard fitting options, the fitting parameter uncertainty is defined [here](https://github.com/ECSHackWeek/impedance.py/blob/6b80b4129115e246adb90f2179d088f3ada5a5ff/impedance/models/circuits/fitting.py#L144). This line is calculating the square root of the diagonal of the covariance matrix, which is the standard deviation (see [here](https://stackoverflow.com/a/52275674)). For the `basinhopping` algorithm, the covariance matrix is first computed, and then the std is computed in the same fashion (line [here](https://github.com/ECSHackWeek/impedance.py/blob/6b80b4129115e246adb90f2179d088f3ada5a5ff/impedance/models/circuits/fitting.py#L193)). Note that the errors of the `basinhopping` algorithm don't always work great.

It would be great to add this to the documentation -- I/someone else can hopefully find the time soon.

petermattia, 2021-08-31:
Actually, it's already documented, see `impedance.models.circuits.fitting.circuit_fit` here:
https://impedancepy.readthedocs.io/en/latest/fitting.html

emrebaburoglu, 2021-09-06:
Hello @petermattia, 
Thank you so much for your explanation! I think i generally understand how it's calculated but still not sure what dataset the standard deviations are calculated for. For example is the standard deviation on R0 calculated from the different R0 values obtained from the successive iterations of the least squares fitting algorithm? I'm not sure what the standard deviations mean physically. 
I apologise if this is all obvious, I'm very new to python and curve fitting in general.

petermattia, 2021-09-07:
Ah, I understand now -- yes, these parameters need to be taken with a grain of salt, as estimates of parameter confidence from nonlinear curve fitting is not fully settled. See here for more info https://stackoverflow.com/a/21844726.

Here are some other references that cover this topic:
- https://www.nature.com/articles/nprot.2009.182 (beginner-friendly, though targeted towards Excel users)
- https://faseb.onlinelibrary.wiley.com/doi/abs/10.1096/fasebj.1.5.3315805
- https://www.cell.com/biophysj/pdf/S0006-3495(83)84281-7.pdf

We should eventually update the documentation here to reflect these caveats.