ECSHackWeek/impedance.py #178: Question about the math of 'complex' fit_type

DreamBoatOve, 2021-09-02:
https://github.com/ECSHackWeek/impedance.py/blob/6b80b4129115e246adb90f2179d088f3ada5a5ff/impedance/validation.py#L247
First, I want to thank Brian Gerwe for answering my last question. your code is very clean and elegant. It is hard to understand at first. After reading the two papers again, I got it at last. It is easy to understand the calculation when fit_type is 'real or imag'. But the calculation is confusing when fit_type is 'complex'. In Line 247-253 , We have Are and Aim and they corresponding Yre = z.re/abs(z), Yim = z.im/abs(z). Shouldn't we just add up Are and Aim --> Are_im, and add up Yre and Yim --> Yre_im, then get elements = pinv(Are_im).T.dot(Yre_im). I am confused about the math here. Could you please share a few sources or keywords, so i can google it myself.

My name is Zhaoyang Zhao and I am a Ph.D. student at the university of science and technology Beijing, China. My email is: b20170427@xs.ustb.edu.cn   Is it appropriate to ask for your email? I want to add you as a co-worker of my project.

DreamBoatOve, 2021-09-02:
I already test the code, it works good. I just don't understand the math

DreamBoatOve, 2021-09-02:
I got it. Sorry to bother
[image]

closed 2021-09-05 by DreamBoatOve

mdmurbach, 2021-09-09:
Awesome, thanks for sharing your work @DreamBoatOve!