ECSHackWeek/impedance.py #183: How to get circuit.parameters_ to display tolerances as well ???

rudybanks, 2021-09-25:
I need to write fit parameter results including the tolerance (+/- x.xx);
circuit.parameters_ allows me to write to a csv file below.

# assign fit parameters to variable data
data = circuit.parameters_

#open csv file - or create file if does not exist
with open('results.csv', 'w', encoding='UTF8') as f:
    writer = csv.writer(f)

    # write the data to file
    writer.writerow(data)

I would like to be able to append these results with the tolerances...

247.68096792261122,1.4830177612597947e-36,3003.119139027305,9.928548425330039e-07

I am trying to important the results in another program after analysis.

Thanks,
Rudy

mdmurbach, 2021-09-26:
Hi Rudy,

Once you have the fit, there are many ways to write the data to a file depending on how you want to store/access it.

For example to export parameters and estimated standard deviations, you can store them in a structured .csv (something like this)

[image]

by using 

```
import pandas as pd

names, units = circuit.get_param_names()

d = {f"{name} [{unit}]": {
        "value": value,
        "std. dev.": confidence
    } for name, unit, value, confidence in zip(names, units, circuit.parameters_, circuit.conf_)}

df = pd.DataFrame().from_dict(d).T
df.to_csv("parameters.csv")
```

or to get the single line format that you mention, something like the following code should work

```
import csv

data = []
for value, conf in zip(circuit.parameters_, circuit.conf_):
    data.extend([value, conf])

with open("./parameters2.csv", "w") as csvfile:
    writer = csv.writer(csvfile)
    writer.writerow(data)
```

which produces one line:

0.016518726102511646,0.00015422764300546203,0.008676550515714202,0.00019127373981880764,3.3214255574929914,0.1895366931922426,0.00538996275465183,0.00020579901137397015,0.06309274229374424,0.0019397399990774887,232.5204246140387,16.226955325866935,0.2195418278135324,0.01754325233570296

rudybanks, 2021-10-04:
Perfect!!!

mdmurbach, 2021-10-19:
Glad this helped!

closed 2021-10-19 by mdmurbach