ECSHackWeek/impedance.py #188: [DATA]

NomaanNabi, 2021-11-25:
* What instrument/software/file type are you requesting support for? * 
* 
* The software used to operate our instrument is the SMaRT v3.3.1 and files are saved in .dat format. 

[Please attach a sample file by dragging and dropping an example file (this file will be public)]

When I try to drag the file it says it is not supported.

BGerwe, 2021-12-02:
Hi,

I believe that's a limitation of the file type GitHub supports. You could share it by:
1. posting a link to another file hosting service (Dropbox or Google Drive)
2. fork this repo, commit/push the file in your forked version, and share the link here

I think 1 is probably easier? But it's up to you

mdmurbach, 2022-09-18:
Hi @NomaanNabi, going to close this issue for now, but feel free to re-open or create a new issue if you're able to figure out a way to share the example data file

closed 2022-09-18 by mdmurbach