ECSHackWeek/impedance.py #189: Impedance ploting issues

Farri-del, 2021-12-02:
Hello. I wanted to fit my supercapacitor impedance data, but there is a problem while plotting the graph. The graph is not drawn correctly, the x-axis is very compressed. How to solve this problem?  here below what did I get. Thanks.

fig, ax = plt.subplots(figsize=(10,10))
ax = circuit.plot(ax, frequencies, Z, kind='nyquist')
ax.tick_params(axis='both', which='major', labelsize=16)
plt.show()

[image]

BGerwe, 2021-12-02:
Hi,

This is a consequence of setting a 1:1 aspect ratio in Nyquist plots:
https://github.com/ECSHackWeek/impedance.py/blob/350a6ea53cb89a6a8916b5f06baada36b996e3ba/impedance/visualization.py#L35-L36

For example, plotting some arbitrary data:
```
_, ax = plt.subplots()
x = [1, 1.1, 1.2]
y = [0, 1, 5]

ax.plot(x, y, 'k-x')
ax.set_aspect('equal')

plt.show()
```
[image]

You can make it more viewer-friendly by manually setting axis limits:
```
_, ax = plt.subplots()

ax.plot(x, y, 'k-x')
ax.set_aspect('equal')
ax.set_xlim(0, 5)
ax.set_ylim(0, 5)

plt.show()
```
[image]

I hope this helps!

Farri-del, 2021-12-03:
Many thanks. Very grateful. Helped!

closed 2021-12-10 by BGerwe