ECSHackWeek/impedance.py #190: Not getting the expected result

BertdeBok, 2022-01-09:
[image]
[image]

What I am getting is:

Name: ZL 0934EP 0024.z
Circuit string: R0-p(R1,CPE)
Fit: True

Initial guesses:
     R0 = 1.00e+02 [Ohm]
     R1 = 1.00e+06 [Ohm]
  CPE_0 = 1.00e-06 [Ohm^-1 sec^a]
  CPE_1 = 9.00e-01 []

Fit parameters:
     R0 = 6.40e+05  (+/- 3.77e-11) [Ohm]
     R1 = 5.67e+08  (+/- 4.11e-12) [Ohm]

  CPE_0 = 1.17e-08  (+/- 2.08e-10) [Ohm^-1 sec^a]
  CPE_1 = 4.99e-01  (+/- 4.31e-03) []

[ZL 0934EP 0024.txt](https://github.com/ECSHackWeek/impedance.py/files/7835153/ZL.0934EP.0024.txt)

BGerwe, 2022-01-10:
It's been a little while since I've used Zview but it looks like only a
selected subset of the data is being fit, whereas with impedance.py the
whole data set is being fit.

Two more things: 1) The negative value of Re is suspicious... 2) Can you
verify the mathematical meanings of Qc-T and Qc-P? I think they can be
found in the manual under the Help tab, but I no longer have access to that
software.

BertdeBok, 2022-01-11:
Dear Brian,

 

Thank you for the quick response.

If I fit over the complete range a have a poor fit but still with a lower Chi_squared

 

 

 

 

 

 

 

 

 

CPE = Constant Phase Element #1        ELEM-CPE

 

       Z = 1 / [T(I*w)^P]

 

       Parameters: CPE-T, CPE-P

 

The CPE is defined by two values, CPE-T and CPE-P. If CPE-P equals 1 than the equation is identical to that of a capacitor. If CPE-P equals 0.5, a 45 degree line is produced on the Complex-Plane graph. When a CPE is placed in parallel to a resistor, a Cole-Element (depressed semi-circle) is produced.

 

Often a CPE is used in a model in place of a capacitor to compensate for non-homogeneity in the system. For example, a rough or porous surface can cause a double-layer capacitance to appear as a constant phase element with a CPE-P value between 0.9 and 1. In fact, a capacitor is actually a constant phase element - one with a constant phase angle of 90 degrees.

 

A CPE with CPE-P value of 0.5 can be used to produce an Infinite Length Warburg element. A Warburg element occurs when charge carrier diffuses through a material. Lower frequencies correspond to diffusion deeper into the material. If the material is thin, low frequencies will penetrate the entire thickness, creating a Finite Length Warburg element. If the material is thick enough so that the lowest frequencies applied do not fully penetrate the layer, it must be interpreted as infinite. The CPE produces the same spectrum as the ‘high frequency’ portion of a Finite Length Warburg when  CPE-T = sqrt(W-T) / W-R.

 

 

 

 

QPE = Constant Phase Element #2        ELEM-CPE

       Z = 1 / [I*wQ)^n]

 

       Parameters: CPE-Q, CPE-n

 

This Constant Phase element has the equation format used by Dr. Bernard Boukamp in his EQUIVCRT program. It will fit exactly the same spectrums as the CPE element, but the Q parameter values will be different from the T parameter values in Constant Phase Element #1.

 

The Q parameter is inside the exponent, while the T value is outside the expoenent. The n and P parameters are identical. The Q and T parameters may translated using the equation    T = (Q)^n

 

 

 

Just for fun I did a fit with both types of Constant Phase Element, not much difference to be seen.

 

 

I agree that a negative electrolyte fysically does not n=make any sence but my total fit is better that way and I do not care about the electrolyte, I am only interested in the Constant Phase Element.

 

 

The sqi-squared as reported by Z-view is much, much lower than with impedance.py and that worries me.

 

Kind regards,

 

Bert de Bok, Technical Manager Transocean Coatings, 

 

Huygensstraat 3

2652 XK

Berkel en Rodenrijs, The Netherlands  <http://www.transocean-coatings.com> www.transocean-coatings.com

Office Tel; +31-10-4134477    Skype; bertdebok

 

Disclaimer.

The information contained in this message is intended to be exclusively for the addressee. No part of this message may be made public, retransmitted, disseminated or for any other use by persons or entities (including the intended recipient) without prior written permission from Transocean.

Transocean cannot be held liable for the contents of this message, nor can any rights be derived from it.

mdmurbach, 2023-02-02:
Going to go ahead and close this one @BertdeBok, but if you have any other questions or issues please feel free to open reopen the issue or post in [the discussion section](https://github.com/ECSHackWeek/impedance.py/discussions/new?category=q-a)

closed 2023-02-02 by mdmurbach