ECSHackWeek/impedance.py #191: Refining fit parameters

fatmaoza, 2022-01-14:
Hi,

I'm trying to fit my data to the following circuit:  'R0-p(R1,C1)-p(R2,C2)-p(R3,C3)'

If I use these initial guesses: [.01, .01, 100, .01, .05, 100, 1] then everything works fine, except that the fit is not really good. However, my idea was to refine the fit by taking the the values from circuit.parameters_fit and using them as new initial guesses. Unfortunately, the fit calculation takes so long, that I always have to break up the calculation (more than 45 min). 

I am wondering, why the fit calculation with the new calculated parameters takes longer than the initial guesses? Should it be not even faster? And is there an opportunity, to speed up the calculations? E.g. even if I only change the parameter of R0 and keep all paramaters constant, the calculation time is still infinte long. Is this problem maybe linked to the "big" circuit, since I habe three RW elements in parallel?

FTHuld, 2022-01-25:
Hi @fatmaoza.

Without an example of your data or code it's hard to know exactly what's going on.
If the fit isn't converging it could be because the fit is stuck in a local minimum. My suspicion however, is that the circuit isn't the best for your data. What happens if you replace some of your capacitors with CPEs? I also notice that your circuit doesn't have a diffusive tail. Does adding a Warburg to your circuit help?

mdmurbach, 2022-09-18:
Going to go ahead and close this issue, but feel free to re-open or create a new issue if you have any other questions @fatmaoza!

closed 2022-09-18 by mdmurbach