ECSHackWeek/impedance.py #192: Set alpha for CPE

fatmaoza, 2022-01-15:
Hi,

I have another question. Currently I'm using the following circuit model: 'R0-p(R1,C1)-p(R2,C2)-p(R3,C3)-p(R4,C4)'

With my initial guess parameters I see, that this model is more or less ok, however, I need to change the alpha values of the CPE elements. I tried out the following:

circuit.fit(frequencies, Z, bounds=[(0.5,0.5,0.5,0.5,0.5,0.5,0.5,0.5,0.5),(1,1,1,1,1,1,1,1,1)])

If I run my script with adding the bounds parameter, I get the error message: ValueError: `x0` is infeasible.

I tried this also with only 5 values per tuple, but then it says: Inconsistent shapes between bounds and `x0`.

Can somoneone help me with this issue? How can I set the alphas of specific CPE elements? And is there an opportunity to also fit the alpha values (Giving some initial guesses and calculating the real alpha value with the fit function)?

Thank you in advance!

FTHuld, 2022-01-25:
Hi @fatmaoza.
bounds takes a tuple, not a list. Try replacing your [ ] with ( ), like this:

`circuit.fit(frequencies, Z, bounds=((0.5,0.5,0.5,0.5,0.5,0.5,0.5,0.5,0.5),(1,1,1,1,1,1,1,1,1)))`

mdmurbach, 2022-09-18:
Thanks for the answer @FTHuld! Hopefully, that solves your issue @fatmaoza, but feel free to re-open or create a new issue if you still have any questions

closed 2022-09-18 by mdmurbach