ECSHackWeek/impedance.py #198: Randles circuit wrong ?

lucasfdvx, 2022-04-12:
When using the Randles class, the circuit is 'R0-p(R1,C1)-Wo1' ('R0-p(R1,CPE1)-Wo1' when CPE=True).
Should the circuit not be 'R0-p(R1-Wo1,C1)' ('R0-p(R1-Wo1,CPE1)' when CPE=True), as described in https://en.wikipedia.org/wiki/Randles_circuit ?
Am I missing something ?

BGerwe, 2022-04-23:
Nope, I think you're totally right. This is embarrassing... but good catch and thank you for opening the issue!

mdmurbach, 2022-04-25:
Fixed in [the v1.4.1 release](https://github.com/ECSHackWeek/impedance.py/releases/tag/v1.4.1)!

You can update with: `pip install --upgrade impedance`

Thanks a bunch @lucasfdvx for reporting the bug and @BGerwe for the quick fix :)

@all-contributors please add @lucasfdvx for bug

closed 2022-04-25 by mdmurbach

des520mond, 2022-06-14:
Dear all, 

I was wondering why the default Randles Circuit 'Randles()' uses open (finite-space) Warburg Impedance 'Wo' rather than semi-infinite Warburg Impedance 'W'. I thought the latter was more appropriate because the Randles Circuit is commonly assumed to be a semi-infinite model, and when the diffusion model is assumed to be finite, only finite-space (Wo) or finite-length (Ws) will be used. Thus, shouldn't the default Randles circuit be defined as 'R0-p(R1-W1,C1)' instead?

Please see [link](http://lacey.se/science/eis/diffusion-impedance/) for the explanations for general, open and short Warburg Impedance.