ECSHackWeek/impedance.py #203: [BUG]

Muneebing, 2022-05-09:
**Describe the bug**
Hello,
I was trying to fit the model by doing this example data set given on the website, as I am a new user so I was trying to get my head around by applying this example but during the Step of circuit Fit with method .fit() I got the error, please look at the screen shot below.
Could you please tell me what the problem here is?
It says TypeError ('length of impedance and frequencies do not match')
[image]

BGerwe, 2022-05-09:
It sounds like the arrays of impedance data and the frequency list are different sizes. Can you show the code used to import the data?

Muneebing, 2022-05-09:
Hello Brian Thanks for the response, please have a look at the code below.
Thanks.
[image]

BGerwe, 2022-05-09:
I'm having trouble reproducing the error you're seeing. But I did notice a small typo in your code. In the first set of parallel elements, they should be separated by a comma (`p(R1,C1)`) instead of a dash (`p(R1-C1)`). 

The error message I got using your original circuit string was "UnboundLocalError: local variable 'split' referenced before assignment", and it looks like your original screenshot also shows "UnboundLocalError" at the top of the traceback.

Can you give this a shot and see if it helps?

BGerwe, 2022-05-10:
The other thing I'm wondering is what dataset you are using. Did you take our [exampleData.csv](https://github.com/ECSHackWeek/impedance.py/blob/9c28fc43bf1b9bf2ee230722defd8b8921ce34f4/data/exampleData.csv) and rename it? Or is the file your own? It seems like the array lengths in the screenshot above don't match up with our example data.

Muneebing, 2022-05-10:
Thanks a lot Brain for catching the Typo, The problem was the wrong definition of parallel RC element. Its resolved now

closed 2022-05-10 by Muneebing