ECSHackWeek/impedance.py #218: Add EIS interpretation resources to README

BGerwe, 2022-09-18:
Add a section to the README providing a few resources for folks who are just getting started with fitting and interpreting impedance data. As maintainers, we don't prioritize helping the community build and tweak equivalent circuits to fit their data, but providing these starting places will allow folks to self-serve.

Some options coming to mind are:
- [Matt Lacey's site](http://www.lacey.se/science/eis/)
- [Gamry primer](https://www.gamry.com/application-notes/EIS/basics-of-electrochemical-impedance-spectroscopy/)
- [Mark Orazem site](http://ww2.che.ufl.edu/orazem/Downloads.htm)
- [Orazem and Tribollet book](https://www.amazon.com/Electrochemical-Impedance-Spectroscopy-Mark-Orazem/dp/0470041404)

nicmazza, 2022-10-27:
Hi, 
I have been reading this book lately, and it has a nice intro to EIS, including how it correlates to other echem techniques, and different circuit elements. Then it goes in depth on how it can be applied to PEM FC.
Since in Matt's website there some resources focused on batteries, and he cites the Boukamp book for solid systems, this book could complement the info there presented. 
I have not read the Orazem book, so I can't judge how much they overlap. 

Xiao-Zi Yuan, Chaojie Song,  Haijiang Wang, Jiujun Zhang
Electrochemical Impedance Spectroscopy in PEM Fuel Cells, Fundamentals and Applications.
Springer
ISBN 978-1-84882-845-2 
e-ISBN 978-1-84882-846-9
DOI 10.1007/978-1-84882-846-9

Yugal0, 2023-12-25:
> Add a section to the README providing a few resources for folks who are just getting started with fitting and interpreting impedance data. As maintainers, we don't prioritize helping the community [...]

The third link to the Orazem's website wont work.