ECSHackWeek/impedance.py #220: [BUG] DepricationWarning NP.complex

SaftMacki, 2022-09-20:
**Describe the bug**
When I try to make a fit I get DepricationWarning.

**To Reproduce**
Steps to reproduce the behavior:
1. cell_model_T.fit(x_values, z_values)
2. DeprecationWarning: `np.complex` is a deprecated alias for the builtin `complex`. To silence this warning, use `complex` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.complex128` here.

**Additional context**
I just removed "np." before complex in line 113 in circuits.py and it seems to work again.

BGerwe, 2022-09-20:
Hi Marcus, thanks for bringing this to our attention. Luckily, since this is just a warning it shouldn't impact the functionality or accuracy of the package even though it may be annoying to see the warnings pop up. You should be able to hide these with the `warnings` package with something like:

```python
import warnings

with warnings.catch_warnings():
   warnings.simplefilter("ignore", DeprecationWarning)
   <other code here>
```

I'll put this in our queue of maintenance tickets, but it would be tremendously helpful (and also be fixed faster) if you could open a pull request to propose a fix. Here are [instructions for contributing](https://github.com/ECSHackWeek/impedance.py/blob/main/CONTRIBUTING.md).

Thanks!

SaftMacki, 2022-09-21:
Sweet. I opened a pull request.

mdmurbach, 2022-09-28:
Closed in https://github.com/ECSHackWeek/impedance.py/pull/221!! Thanks again @SaftMacki

closed 2022-09-28 by mdmurbach