ECSHackWeek/impedance.py #232: [BUG] (not sure it is a bug)

flamariond, 2022-12-12:
**Describe the bug**
Fitting seems to be  inadequate for a p(R1,C1) circuit with large values of R (eg 1e+9) e small values of C (eg 1e-12)

**To Reproduce**
Steps to reproduce the behavior:
1. I have a synthetic data for a p(R,C) circuit in the frequency range of 10^5 Hz to 1 Hz
2. impedance.py does not fit it. It seems to hold constant one of the parameters.

**Expected behavior**
Since it is a synthetic data, fit should be perfect.

**Additional context**
I have been using this excellent package for a couple of days, and found this difficulty trying to fit data obtained from a painted metallic surface on NaCl 3.5% (it is a corrosion study), with these high values of R and small values of C.
Congratulations for this excellent software.

mdmurbach, 2023-01-28:
Hi @flamariond, thanks! Would you be able to share the code you have to reproduce this?

adrianusler, 2024-08-16:
I also had a problem with fitting impedance data when the parameters were supposed to be very small $<10^{-10}$. This was related to a bug in scipy.optimize at the time: https://github.com/scipy/scipy/issues/18793#issue-1781188767
Could this be related to your problem, @flamariond? Which scipy version do you use?

BGerwe, 2026-02-27:
Sorry we never got back to you about the issue. Hopefully you were able to resolve it, but reopen this if you're still having trouble.