ECSHackWeek/impedance.py #234: Need help fitting data, fitted parameters off shoot.

kevinsmia1939, 2022-12-25:
Hello,

I am trying to fit a parameter to this data gather from vanadium redox cell.

Remove .txt from file extension to get .par, versastudio file
[EIS_V3.par.txt](https://github.com/ECSHackWeek/impedance.py/files/10299796/EIS_V3.par.txt)
[image]

The issue occur when I add Warberg element or use Randles circuit.
I was able to fitted the first semi circle with initial guess of R0, R1, and CPE quite well.
But the tail does not give me a good fit and the value blows up.

Any idea?
Thank you.
```
from impedance import preprocessing
from impedance.models.circuits import CustomCircuit
import matplotlib.pyplot as plt
from impedance.visualization import plot_nyquist
import numpy as np
from impedance.models.circuits import Randles, CustomCircuit
eis_file = 'EIS_V3.par'
frequencies, z = preprocessing.readVersaStudio(eis_file)
frequencies, z = preprocessing.ignoreBelowX(frequencies, z)

fig, ax = plt.subplots(figsize=(10,10))
# circuit = CustomCircuit('R0-p(R1-W1,CPE1)', initial_guess=[1.2],constants={'R1':0.35, 'CPE1_0':0.7, 'CPE1_1':0.6,'W1':0.07})
# circuit = CustomCircuit('R0-p(R1-W1,CPE1)', initial_guess=[1.2,0.35,0.7,0.6,0.07])
circuit = Randles(initial_guess=[1.2,0.35,0.7,0.6,0.07,0.5],CPE=True)
f_pred = np.logspace(6,-2)
circuit.fit(frequencies, z)
circuit_fit = circuit.predict(f_pred)
plot_nyquist(ax, z,fmt='.')
plot_nyquist(ax, circuit_fit, fmt='--')
plt.show()
print(circuit)
```

oslopanda, 2023-01-04:
It most probably not a bug.

if you print your frequencies, it ranges from  10E5 to 10E2
so it is not the fitting problem, but the range you have chosen to predict.

try the following:
```
from impedance import preprocessing
from impedance.models.circuits import CustomCircuit
import matplotlib.pyplot as plt
from impedance.visualization import plot_nyquist
import numpy as np
from impedance.models.circuits import Randles, CustomCircuit
eis_file = 'EIS_V3.par'
frequencies, z = preprocessing.readVersaStudio(eis_file)
#frequencies, z = preprocessing.ignoreBelowX(frequencies, z)

fig, ax = plt.subplots(figsize=(10,10))

#circuit = CustomCircuit('L0-R0-p(R1-Wo1,CPE1)', initial_guess=[0.000001,1.2,0.35,900,1,0.6,0.9])
circuit = Randles(initial_guess=[1.2,0.35,90,1,0.6,0.9],CPE=True)
f_pred = np.logspace(5,2)
circuit.fit(frequencies, z)
circuit_fit = circuit.predict(f_pred)
plot_nyquist(ax, z,fmt='.')
plot_nyquist(ax, circuit_fit, fmt='--')
plt.show()
print(circuit)
![download (2)](https://user-images.githubusercontent.com/33810430/210560680-99a0df94-292c-40e1-bb80-2bf4a498fb06.png)

print (frequencies)
```
[image]

The fitting is not good probably due to the deep slop of the W element high than 45 degree.

mdmurbach, 2023-02-02:
Thanks for following up @oslopanda! Going to close this issue, but feel free to open another chat in [the discussion section](https://github.com/ECSHackWeek/impedance.py/discussions/new?category=q-a) if you still are having issues or have questions @kevinsmia1939

closed 2023-02-02 by mdmurbach