ECSHackWeek/impedance.py #236: [BUG] AttributeError: module 'numpy' has no attribute 'complex'

pililac, 2023-01-02:
Today I updated my mini-forge environment. My impedance script ran well beforehand and failed afterward. I suspect the updated numpy package. Was numpy.dtype numpy.complex removed/changed?

Here is the error log from my jupyter notebook:

````
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[10], line 77
     74 bounds_CPE3_0 = [8e-3, 3.8e-1]
     75 bounds_CPE3_1 = [0.2, 0.7]
---> 77 circuit.fit(
     78     frequencies,
     79     Z,
     80     bounds=(
     81         [
     82             bounds_R0[0],
     83             bounds_R1[0],
     84             bounds_C1[0],
     85             bounds_R2[0],
     86             bounds_CPE2_0[0],
     87             bounds_CPE2_1[0],
     88             bounds_R3[0],
     89             bounds_Wo3_0[0],
     90             bounds_Wo3_1[0],
     91             bounds_CPE3_0[0],
     92             bounds_CPE3_1[0],
     93         ],
     94         [
     95             bounds_R0[1],
     96             bounds_R1[1],
     97             bounds_C1[1],
     98             bounds_R2[1],
     99             bounds_CPE2_0[1],
    100             bounds_CPE2_1[1],
    101             bounds_R3[1],
    102             bounds_Wo3_0[1],
    103             bounds_Wo3_1[1],
    104             bounds_CPE3_0[1],
    105             bounds_CPE3_1[1],
    106         ],
    107     ),
    108     # global_opt=True,
    109     # weight_by_modulus=True,
    110 )
    112 loop_results = pd.DataFrame(
    113     [
    114         {
   (...)
    118     ]
    119 )
    120 for idx, param in enumerate(circuit.get_param_names()[0]):

File C:\other_software\miniforge3\envs\data_analysis\lib\site-packages\impedance\models\circuits\circuits.py:107, in BaseCircuit.fit(self, frequencies, impedance, bounds, weight_by_modulus, **kwargs)
    105 if not isinstance(impedance, np.ndarray):
    106     raise TypeError('impedance is not of type np.ndarray')
--> 107 if impedance.dtype != np.complex:
    108     raise TypeError('impedance array should have a complex ' +
    109                     f'dtype (currently {impedance.dtype})')
    110 if len(frequencies) != len(impedance):

File C:\other_software\miniforge3\envs\data_analysis\lib\site-packages\numpy\__init__.py:284, in __getattr__(attr)
    281     from .testing import Tester
    282     return Tester
--> 284 raise AttributeError("module {!r} has no attribute "
    285                      "{!r}".format(__name__, attr))

AttributeError: module 'numpy' has no attribute 'complex'

````

[... 235 lines cut ...]
vc                        14.3                 h3d8a991_9    conda-forge
vs2015_runtime            14.32.31332          h1d6e394_9    conda-forge
wcwidth                   0.2.5              pyh9f0ad1d_2    conda-forge
webencodings              0.5.1                      py_1    conda-forge
websocket-client          1.4.2              pyhd8ed1ab_0    conda-forge
whatthepatch              1.0.3              pyhd8ed1ab_0    conda-forge
wheel                     0.38.4             pyhd8ed1ab_0    conda-forge
widgetsnbextension        4.0.5              pyhd8ed1ab_0    conda-forge
win_inet_pton             1.1.0              pyhd8ed1ab_6    conda-forge
winpty                    0.4.3                         4    conda-forge
wrapt                     1.14.1          py310h8d17308_1    conda-forge
xlsxwriter                3.0.5              pyhd8ed1ab_0    conda-forge
xorg-libxau               1.0.9                hcd874cb_0    conda-forge
xorg-libxdmcp             1.1.3                hcd874cb_0    conda-forge
xz                        5.2.6                h8d14728_0    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
yapf                      0.32.0             pyhd8ed1ab_0    conda-forge
zeromq                    4.3.4                h0e60522_1    conda-forge
zipp                      3.11.0             pyhd8ed1ab_0    conda-forge
zstandard                 0.19.0          py310h0009e47_1    conda-forge
zstd                      1.5.2                h7755175_4    conda-forge

</details>

````

oslopanda, 2023-01-03:
It is because that the channel release is not updated (the code here is updated)
So, one solution before the release updated:
go to your directory:
C:\other_software\miniforge3\envs\data_analysis\lib\site-packages\impedance\models\circuits\circuits.py
from line 107
Comment out the lines 107, 108 and 109:
# if impedance.dtype != np.complex:
 #raise TypeError('impedance array should have a complex ' +
# f'dtype (currently {impedance.dtype})')

save and it will temperately fix your problem.

or you could copy the code from here to your circuits.py.

mdmurbach, 2023-01-28:
Thanks for reaching out! Yeah, as @oslopanda mentioned in #235 the issue here was that the fixed version never got updated on PyPI. I've just released [v1.4.2](https://pypi.org/project/impedance/1.4.2/) and you should be able to update (`pip install impedance --upgrade`) and it should work without the workaround.

@all-contributors please add @pililac for bug report

closed 2023-01-28 by mdmurbach