ECSHackWeek/impedance.py #239: Help needed for using miniconda and impedance.py modules

chuckm990, 2023-01-25:
Hi people, I'm kinda new to python, and I wanted to use the EIS analysis suite that was developed by the ECSHackWeek guys. I am having a problem with data import, I am trying to follow all the instructions as listed here: https://impedancepy.readthedocs.io/en/latest/getting-started.html#step-2-import-your-data, though I can't seem to work my way through it and I always get this error when trying to call the "from impedance import preprocessing" I get this message: 'is not recognized as an internal or external command, operable program or batch file.' I am so annoyed, I have enabled systems environment variables (I read those on another troubleshooting blogs), but I may still have a problem there. I am using the latest miniconda python application on Windows 10.

Many thanks.

closed 2023-01-26 by chuckm990

mdmurbach, 2023-01-28:
Hopefully you closed this because you figured it out, but feel free to reopen if you're still having any issues

chuckm990, 2023-01-29:
Dear Matt,

Thank you so much, the software is very good, I used it for 3 days, and the
Lin-KK algorithm implemented for the validation of expermental data has
proven really useful. As for the data retrieving, could you help me with
the command? So far, it's about the only thing I have yet to learn how to
do.

Another question, are there any seminars (or webinars) or courses on how to
use this software and interpret the results? I am deeply interested in any
of those, since my work is related to EIS as an experimental researcher.

Many thanks,
Best regards,

cm

mdmurbach, 2023-01-29:
Hi @chuckm990,

We don't have any recorded seminars and the best place to start is usually the [Getting Started](https://impedancepy.readthedocs.io/en/latest/getting-started.html) documentation. I find that using conda and Jupyter Hub is typically the easiest way to get started. For more info on Python and Jupyter Lab, you might also be interested in some of the tutorials from the [ECS Hackweek 2019](https://ecshackweek.github.io/tutorial/).

Matt