ECSHackWeek/impedance.py #245: [BUG] .mpr file related to PEIS measurment (aquisited from a VPM3 biologic pootentiostat) is not supporting. error: "Column ID 17 after column Analog IN 1/V is unknown"

Mehdisoley, 2023-02-01:
I wanted to put my data (PEIS data) aquisited from a VPM3 biologic pootentiostat into a pandas library. 
This is my code:

from galvani import BioLogic
import pandas as pd
mpr_file = BioLogic.MPRfile("Cr-1-polveri_di_Grafite_01_PEIS_C01.mpr")
df = pd.DataFrame(mpr_file.data)
but it return me this error.
NotImplementedError                       Traceback (most recent call last)
Input In [11], in <cell line: 4>()
      1 from galvani import BioLogic
      2 import pandas as pd
----> 4 mpr_file = BioLogic.MPRfile("Cr-1-polveri_di_Grafite_01_PEIS_C01.mpr")
      5 df = pd.DataFrame(mpr_file.data)

File D:\Anaconda\lib\site-packages\galvani\BioLogic.py:413, in MPRfile.__init__(self, file_or_path)
    408     raise ValueError("Unrecognised version for data module: %d" %
    409                      data_module['version'])
    411 assert(not any(remaining_headers))
--> 413 self.dtype, self.flags_dict = VMPdata_dtype_from_colIDs(column_types)
    414 self.data = np.frombuffer(main_data, dtype=self.dtype)
    415 assert(self.data.shape[0] == n_data_points)

File D:\Anaconda\lib\site-packages\galvani\BioLogic.py:310, in VMPdata_dtype_from_colIDs(colIDs)
    308         type_list.append((unique_field_name, field_type))
    309     else:
--> 310         raise NotImplementedError("Column ID {cid} after column {prev} "
    311                                   "is unknown"
    312                                   .format(cid=colID,
    313                                           prev=type_list[-1][0]))
    314 return np.dtype(type_list), flags_dict

**NotImplementedError: Column ID 17 after column Analog IN 1/V is unknown**

BGerwe, 2023-02-02:
Hi there, thanks for opening the issue. Respectfully, it seems like this error is coming from the `galvani` package and not `impedance`.  We've played with supporting the .mpr file type (#115), but got stalled with confusion over proper licensing procedure.

If you export the file to a .mpt in the BioLogic software, you'll be able to read it in with `impedance.preprocessing.readBioLogic()`

Mehdisoley, 2023-02-02:
Thank you for your answer.
yes when I convert it to txt file it is working. however as I am dealing with a lot of file it would be better to work with original data.
Is there any Forum or Room about galvani so I can publish my issue there?

closed 2023-02-20 by mdmurbach