ECSHackWeek/impedance.py #253: [BUG] Formula wrong for TLMQ in elements.py?

pililac, 2023-02-20:
The formula for circuit element TLMQ in 

J. Landesfeind et al.,
    Journal of The Electrochemical Society, 163 (7) A1373-A1387 (2016)
    `doi: 10.1016/10.1149/2.1141607jes
    <http://doi.org/10.1149/2.1141607jes>`_.

is given by:

[image]

The formula in ```def TLMQ(p, f)``` in elements.py in line 335 is

```
Z = np.sqrt(Rion / Zs) / np.tanh(np.sqrt(Rion * Zs))
```

Translated from ```tanh``` to ```coth``` this is:

```
Z = np.sqrt(Rion / Zs) * np.coth(np.sqrt(Rion * Zs))
```

But I think it should be

```
Z = np.sqrt(Rion * Zs) * np.coth(np.sqrt(Rion / Zs))
```
with the Terms ```(Rion * Zs)``` and ```(Rion / Zs)``` switched between the both ```np.sqrt(...)```-parts to correspond to the formula in the paper. Am I wrong?

mdmurbach, 2023-02-20:
Hi @pililac, thanks for reaching out! I think the issue here isn't in the final equation, but rather in how we are defining the intermediate, Zs. It is defined as `1/(Q(iw)^gamma)` in the paper and in the code `Zs = Q(iw)^gamma` so the final equation in the code and paper end up matching with the "switched" `*` and `/`. Even though this means that the output of the TLMQ element is technically correct, I do think it's worth changing since it's confusing to use Zs incorrectly in the code.

mdmurbach, 2023-02-20:
Actually, as I try to fix this and rerun the tests I do think it matters since `Qs` is not being raised to the `-gamma` so it does change the fit parameters.

pililac, 2023-02-21:
Thanks for looking into the formulas, and sorry for the confusion. I think the formula was right, I have overseen the reciprocal definition of ```Zs```. This compensates the raise to ```-1``` of both ```Qs``` and ```(j * omega) ** gamma``` used in the paper.

PS: Is there a reason for using ```... / np.tanh``` instead of ```... * np.coth``` in a lot of equations? Performance? It is sometimes hard to read the equations and compare them to the formulas in the docstrings. Would it help to increase readability by transforming the formulas? Or do you have some kind of best practice you are follwoing?

BGerwe, 2023-02-22:
> Actually, as I try to fix this and rerun the tests I do think it matters since Qs is not being raised to the -gamma so it does change the fit parameters

Hmm, maybe I'm missing something, but I do think the output of TLMQ in code matches the paper, though I agree we should refactor to match the paper so folks don't need to waste brain cycles thinking about this in the future. Let's just convert the paper to `coth` for better comparison. $coth(x) = \frac{1}{tanh(x)}$, right? So the equation from the paper should be

# $Z_{el} = \sqrt{\frac{R_{ion}}{Q_s(j\omega)^{\gamma}}} \frac{1}{tanh(\sqrt{Q_s(j\omega)^{\gamma}R_{ion}})}$

Then if we use our definition of $Z_s$ from the code:

# $Z_s = Q_s(j\omega)^{\gamma}$

And substitute that into the above we get:

# $Z_{el} = \sqrt{\frac{R_{ion}}{Z_s}} \frac{1}{tanh(\sqrt{Z_sR_{ion}})}$

Which is how we have $Z_el$ defined in code: `Z = np.sqrt(Rion / Zs) / np.tanh(np.sqrt(Rion * Zs))`

So, I *think* it's technically correct, but we definitely should refactor for clarity.

BGerwe, 2023-02-23:
@pililac 
> PS: Is there a reason for using ... / np.tanh instead of ... * np.coth in a lot of equations? Performance? It is sometimes hard to read the equations and compare them to the formulas in the docstrings. Would it help to increase readability by transforming the formulas? Or do you have some kind of best practice you are following?

I didn't know this until looking into it right now, but `numpy` doesn't have an implementation of coth. So, we're forced to adjust the equations to work with `np.tanh`

commit cfacda2f (2023-06-24): modifies equation for TLMQ element, fixes #253 (#255)

* modifies equation for TLMQ element, fixes #253

* fix equation for Zs
closed 2023-06-24 by mdmurbach