ECSHackWeek/impedance.py #262: [Element]Define a custom circuit element？

2410337637, 2023-04-11:
I've been using impedance.py for a while, and it's very convenient for EIS fitting, but as my research progresses, I'd like to build some special circuit elements, and I was wondering if there's a tutorial here for customizing a circuit element based on specific impedance formulas.

BGerwe, 2023-04-11:
Hi,

Thanks for using impedance.py! Defining custom elements is a recent feature, introduced in the latest release, v1.5.0. We have a basic usage example in the [fitting notebook of our documentation](https://impedancepy.readthedocs.io/en/latest/examples/fitting_example.html#Custom-circuit-elements-using-the-@element-decorator). Hopefully that's enough to show you the usage pattern so you can apply it to your specific case.

2410337637, 2023-04-12:
@BGerwe 
Oh, thanks for the reply! I just noticed that a tutorial has been updated here, great, I have successfully done the custom component.

closed 2023-04-12 by BGerwe